PX4/PX4-Autopilot #6664: 2 bugs in TECS

Jiangyou0811, 2017-02-24:
in tecs.cpp
1. in line 110, it should be
`temp = rotMat(0, 2) * CONSTANTS_ONE_G + accel_body(0);`
rather than 
`temp = rotMat(2, 0) * CONSTANTS_ONE_G + accel_body(0);`

because the equation should be rotMat * [0;0;G], which means the element in 1st row 3rd column is enrolled. 

2. From line 283 to 285
`_hgt_dem_adj = 0.1f * _hgt_dem + 0.9f * _hgt_dem_adj_last;
_hgt_dem_adj_last = _hgt_dem_adj;
_hgt_rate_dem = (_hgt_dem_adj - state) * _heightrate_p + _heightrate_ff * (_hgt_dem_adj - _hgt_dem_adj_last) / _DT;
`
line 284 and line 285 must be exchanged. Otherwise the heightrate_ff won't have effect. The derivative is always zero.

LorenzMeier, 2017-02-24:
Thanks! @dagar @tumbili Can you investigate?

tumbili, 2017-02-25:
Yes, the ff can't work. I need to check the sign of the rotation. Will do a PR.

dagar, 2017-02-26:
Work in progress PR #6675

tumbili, 2017-02-27:
@Jiangyou0811 The rotation matrix used here is one that transforms vector coordinates from body frame to earth frame. That means if you pre-multiply a vector in body frame with the matrix you will get its representation in earth frame, see e.g. here https://github.com/PX4/Firmware/blob/master/src/modules/fw_pos_control_l1/fw_pos_control_l1_main.cpp#L1300
 In tecs we want to transform the earth frame gravity vector to body frame so we need to transpose the matrix before pre-multiplying. That's the reason why it should be rotMat(2,0) and not rotMat(0,2).
Do you agree?

Jiangyou0811, 2017-02-27:
@tumbili Yes, you are correct. I didn't notice a body frame to earth frame matrix is is used. I though this matrix is directly taken from AHRS which should be a earth frame to body frame.

tumbili, 2017-02-27:
@Jiangyou0811 Cool, thanks for the feed-forward find!

PR #6675 (2017-04-05): TECS fix height demand change
 - fixes #6664

Need to do more testing/comparison. TECS might be slightly easier to tune if we expose a few more things over mavlink.
closed 2017-04-05 by LorenzMeier