PX4/PX4-Autopilot #6668: How to fix the problem ' fatal: Not a git repository: XXXX'

newmsg, 2017-02-25:
Hello, 

    I used the cmd "git submodule update --init --recursive" to update all submodules of px4 like this:

$ git submodule update --init --recursive
Submodule 'NuttX' () registered for path 'NuttX'
Submodule 'Tools/gencpp' () registered for path 'Tools/gencpp'
Submodule 'Tools/genmsg' () registered for path 'Tools/genmsg'
Submodule 'Tools/jMAVSim' () registered for path 'Tools/jMAVSim'
Submodule 'Tools/sitl_gazebo' () registered for path 'Tools/sitl_gazebo'
Submodule 'cmake/cmake_hexagon' () registered for path 'cmake/cmake_hexagon'
Submodule 'mavlink/include/mavlink/v1.0' () registered for path 'mavlink/include/mavlink/v1.0'
Submodule 'mavlink/include/mavlink/v2.0' () registered for path 'mavlink/include/mavlink/v2.0'
Submodule 'src/drivers/gps/devices' () registered for path 'src/drivers/gps/devices'
Submodule 'src/lib/DriverFramework' () registered for path 'src/lib/DriverFramework'
Submodule 'src/lib/ecl' () registered for path 'src/lib/ecl'
Submodule 'src/lib/matrix' () registered for path 'src/lib/matrix'
Submodule 'src/modules/uavcan/libuavcan' () registered for path 'src/modules/uavcan/libuavcan'
Submodule 'unittests/googletest' () registered for path 'unittests/googletest'
Submodule 'jMAVlib' (https://github.com/PX4/jMAVSim.git) registered for path 'jMAVlib'
Submodule 'external/OpticalFlow' (https://github.com/PX4/sitl_gazebo.git) registered for path 'external/OpticalFlow'
Submodule 'external/klt_feature_tracker' (https://github.com/PX4/OpticalFlow.git) registered for path 'external/klt_feature_tracker'
Submodule 'cmake_hexagon' (https://github.com/PX4/DriverFramework.git) registered for path 'cmake/cmake_hexagon'
Submodule 'dspal' (https://github.com/PX4/DriverFramework.git) registered for path 'dspal'
fatal: Not a git repository: j:/PX4_SRC/Firmware/.git/modules/src/lib/DriverFramework/modules/cmake/cmake_hexagon
Unable to find current revision in submodule path 'cmake/cmake_hexagon'
Submodule 'matrix' (https://github.com/PX4/ecl.git) registered for path 'matrix'
Submodule 'dsdl' (https://github.com/UAVCAN/libuavcan.git) registered for path 'dsdl'
Submodule 'libuavcan/dsdl_compiler/pyuavcan' (https://github.com/UAVCAN/libuavcan.git) registered for path 'libuavcan/dsdl_compiler/pyuavcan'
Submodule 'dsdl' (https://github.com/UAVCAN/pyuavcan) registered for path 'dsdl'
Failed to recurse into submodule path 'src/lib/DriverFramework'

I found the following error:
fatal: Not a git repository: j:/PX4_SRC/Firmware/.git/modules/src/lib/DriverFramework/modules/cmake/cmake_hexagon
Unable to find current revision in submodule path 'cmake/cmake_hexagon'

However,  the absolute path is "J:\PX4_SRC\Firmware\cmake\cmake_hexagon"
instead of "src/lib/DriverFramework/modules/cmake/cmake_hexagon"

How to fix it? Thank you in advance!

any help would be appreciated!

dagar, 2017-02-26:
Try `make submodulesclean`

dagar, 2017-03-04:
Looking at improving Windows 10 support here - https://github.com/PX4/Firmware/issues/6697

closed 2017-03-04 by dagar