PX4/PX4-Autopilot #6672: cannot run make posix_sitl_default gazebo

tonyvanhain, 2017-02-25:
I think the newest update to the firmware have a bug in it where i cannot run the package, and it hangs my laptop everytime I want to run the command make posix_sitl_default gazebo. Does anyone else have this problem as well?

mhkabir, 2017-02-25:
Where does it hang?

dagar, 2017-02-25:
@tonyvanhain please share the console output or provide a screenshot.
Another quick thing to try is running without the gui.

    HEADLESS=1 make posix_sitl_default gazebo

tonyvanhain, 2017-02-26:
@dagar it still wont run properly using the command you given unfortunately

@dagar @mhkabir Here is the console part where it hangs:
[image], 

I can still run the simulation using make posix_sitl_default jmavsim, but I can't run it now using gazebo

tonyvanhain, 2017-02-26:
[ 52%] Building CXX object CMakeFiles/LiftDragPlugin.dir/src/liftdrag_plugin/liftdrag_plugin.cpp.o
[ 54%] Building CXX object CMakeFiles/gazebo_geotagged_images_plugin.dir/src/gazebo_geotagged_images_plugin.cpp.o
[ 56%] Building CXX object CMakeFiles/rotors_gazebo_wind_plugin.dir/src/gazebo_wind_plugin.cpp.o
[ 58%] Building CXX object CMakeFiles/rotors_gazebo_controller_interface.dir/src/gazebo_controller_interface.cpp.o

After checking, the error seems to be caused when trying to build the files above, anyone can help with it?Building the files above causes my laptop to hang

mhkabir, 2017-02-26:
Do you have enough RAM on your computer? I've seen similar odd behavior
(hangs, freezes, compiler crashes) on systems with <= 4GB RAM.

As a workaround, you can try calling the make target with -j1 to limit
parallel compilation.

tonyvanhain, 2017-02-26:
@mhkabir  My laptop only uses 4GB, the weird thing is that a week ago I was able to run that command using the laptop that I am using right now, but then I deleted the src folder as a whole and then re-do the steps on the px4 site(https://dev.px4.io/starting-building.html ), particularly the steps where it downloads the firmware from the repo. Can that be the issue?

tonyvanhain, 2017-02-27:
I already tried to reinstall ubuntu as a whole, and the problem still persist, can anyone help or also had this situation before?

tonyvanhain, 2017-02-27:
this is what i got when I try to run make posix_sitl_default gazebo
[  0%] Built target git_matrix
[  1%] Built target git_mavlink
[  1%] Built target git_uavcan
[  1%] Built target prebuild_targets
[  1%] Built target git_genmsg
[  1%] Built target git_gencpp
[  1%] Built target git_jmavsim
[  3%] Built target df_driver_framework
[  3%] Built target git_ecl
[  4%] Built target git_driverframework
[  4%] Built target git_gazebo
[  5%] Built target xml_gen
[  5%] Built target mixer_gen
[  5%] Built target ver_gen
[  5%] Performing build step for 'sitl_gazebo'
-- install-prefix: /usr
[ 24%] Built target msg_gen
[ 24%] Built target generate_topic_listener
[ 24%] Built target platforms__common
[ 24%] Built target drivers__boards__sitl
[ 25%] Built target drivers__camera_trigger
[ 30%] Built target systemcmds__tests
[ 31%] Built target drivers__gps
[ 33%] Built target drivers__device
[ 33%] Built target drivers__pwm_out_sim
[ 34%] Built target drivers__vmount
[ 35%] Built target platforms__posix__drivers__accelsim
[ 35%] Built target platforms__posix__drivers__adcsim
[ 35%] Built target platforms__posix__drivers__airspeedsim
[ 36%] Built target platforms__posix__drivers__barosim
[ 36%] Built target platforms__posix__drivers__gpssim
[ 37%] Built target platforms__posix__drivers__gyrosim
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   thread
--   signals
--   system
--   filesystem
--   program_options
--   regex
--   iostreams
--   date_time
--   chrono
--   atomic
[ 37%] Built target platforms__posix__drivers__rgbledsim
[ 37%] Built target platforms__posix__drivers__ledsim
[ 38%] Built target platforms__posix__drivers__tonealrmsim
-- Boost version: 1.58.0
-- Looking for OGRE...
-- Found Ogre Ghadamon (1.9.0)
-- Found OGRE: optimized;/usr/lib/x86_64-linux-gnu/libOgreMain.so;debug;/usr/lib/x86_64-linux-gnu/libOgreMain.so
[ 42%] Built target platforms__posix__work_queue
-- Looking for OGRE_Paging...
-- Found OGRE_Paging: optimized;/usr/lib/x86_64-linux-gnu/libOgrePaging.so;debug;/usr/lib/x86_64-linux-gnu/libOgrePaging.so
-- Looking for OGRE_Terrain...
-- Found OGRE_Terrain: optimized;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so;debug;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so
-- Looking for OGRE_Property...
-- Found OGRE_Property: optimized;/usr/lib/x86_64-linux-gnu/libOgreProperty.so;debug;/usr/lib/x86_64-linux-gnu/libOgreProperty.so
-- Looking for OGRE_RTShaderSystem...
-- Found OGRE_RTShaderSystem: optimized;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so;debug;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so
-- Looking for OGRE_Volume...
-- Found OGRE_Volume: optimized;/usr/lib/x86_64-linux-gnu/libOgreVolume.so;debug;/usr/lib/x86_64-linux-gnu/libOgreVolume.so
-- Looking for OGRE_Overlay...
[ 42%] Built target systemcmds__esc_calib
-- Found OGRE_Overlay: optimized;/usr/lib/x86_64-linux-gnu/libOgreOverlay.so;debug;/usr/lib/x86_64-linux-gnu/libOgreOverlay.so
[ 43%] Built target platforms__posix__px4_layer
[ 44%] Built target systemcmds__perf
[ 45%] Built target systemcmds__mixer
[ 45%] Built target systemcmds__param
[... 128 lines cut ...]
make[7]: *** Waiting for unfinished jobs....
make[7]: *** wait: No child processes.  Stop.
make[7]: *** wait: No child processes.  Stop.
make[7]: *** Waiting for unfinished jobs....
make[7]: *** wait: No child processes.  Stop.
CMakeFiles/Makefile2:141: recipe for target 'CMakeFiles/rotors_gazebo_wind_plugin.dir/all' failed
make[6]: *** [CMakeFiles/rotors_gazebo_wind_plugin.dir/all] Error 2
CMakeFiles/Makefile2:104: recipe for target 'CMakeFiles/gazebo_geotagged_images_plugin.dir/all' failed
make[6]: *** [CMakeFiles/gazebo_geotagged_images_plugin.dir/all] Error 2
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/LiftDragPlugin.dir/all' failed
make[6]: *** [CMakeFiles/LiftDragPlugin.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make[5]: *** [all] Interrupt
src/firmware/posix/CMakeFiles/sitl_gazebo.dir/build.make:110: recipe for target 'external/Stamp/sitl_gazebo/sitl_gazebo-build' failed
make[4]: *** [external/Stamp/sitl_gazebo/sitl_gazebo-build] Interrupt
CMakeFiles/Makefile2:7861: recipe for target 'src/firmware/posix/CMakeFiles/sitl_gazebo.dir/all' failed
make[3]: *** [src/firmware/posix/CMakeFiles/sitl_gazebo.dir/all] Interrupt
CMakeFiles/Makefile2:11576: recipe for target 'src/firmware/posix/CMakeFiles/gazebo.dir/rule' failed
make[2]: *** [src/firmware/posix/CMakeFiles/gazebo.dir/rule] Interrupt
Makefile:3978: recipe for target 'gazebo' failed
make[1]: *** [gazebo] Interrupt
Makefile:139: recipe for target 'posix_sitl_default' failed
make: *** [posix_sitl_default] Interrupt

I interrupt the process because if I don't the hang will begin afterward and I have to force shutdown my laptop

tonyvanhain, 2017-02-28:
I have upgraded my RAM, and the issues seems to have been solved. Thank you @mhkabir for the suggestion.

closed 2017-02-28 by tonyvanhain