PX4/PX4-Autopilot #6681: Descending slew rate overshot

matanhavi, 2017-02-27:
When pause the drone during descending, its keep descending due to target altitude slew rate (seems to).
This scenario might cause significant altitude over-shots.
Produced on SITL (iris, ekf2)
1)	Hovering at 100 m
2)	Start descending (from 100m to 50m)
3)	Pause the drone when it’s at 70m, (target altitude is already at 50m)  
4)	Target altitude start rising back, (Bug is here, it should be fixed to current 70m) yet its lower than current altitude so the drone is keep descending
5)	Only when target altitude reaches the current altitude, the drone will start climbing again 
6)	In this scenario, the drone has already 10m overshot. (it can easily changed to higher overshoots)
[image]

LorenzMeier, 2017-02-28:
@Stifael can you have a look?

Stifael, 2017-02-28:
@lorenz: thats the exact same issue that I showed you yesterday with the delay  when flying in the xy direction and demanding a fast switch in the opposite direction. The root cause is the slewrate indeed.
@matanhavi: even though your analysis is correct, I never saw it overshooting as much as you showed on the plot. What is the max acceleration you used on the upward direction? 
I actually was planing to get rid of the slewrate in xy and test it tomorrow. However, the slewrate in z is especially helpful when flying upwards and then doing fast switch downwards. The slewrate might not be needed when doing the opposite, flying downwards and then switch back to upward flight. But need to verify this first.
In my opinion the main issue with slewrate is that the setpoint changes in a feedforward fashion, not considering the actual state of the vehicle. The correct way to limit acceleration would be to do it at the output of the velocity controller (this requires though to have acceleration instead of normalized thrust as output)

LorenzMeier, 2017-02-28:
If the slew rate limiting is producing an offset then it must be a subtle oversight in how its applied.

@Stifael The test yesterday was with incorrectly set angle limits and the system never actually ran into the slew rate limit. So I'm not sure how useful that was.

LorenzMeier, 2017-02-28:
@MaEtUgR Has the details on that.

Stifael, 2017-02-28:
I have looked intovthis issue today together with @MaEtUgR and you can see this behavior from the plots. Its not about hitting any limits, its about how quickly the setpoint can change. The reason why we also saw it yesterday in contrast to a week ago was the increase in max velocity: at tigher velocity but same slewrate limit (which is set by the acc_max), the velocity setpoint needs more time to switch direction.

Stifael, 2017-02-28:
I will upload some plots tomorrow where one can see that the velocity setpoint lacks behind the user input. While the user demands directional change, the velocity setpoint remains in the opposite direction and the velocity setpoint direction change occurs with a clear delay (its also to note that windup issue can be excluded since it happens with zero integration as well). It can also be reproduced in sitl and looking at the velocity setpoint change during switch transition.

Stifael, 2017-03-01:
@matanhavi: about point 4. im confident to know how to fix that bug. I will
do a pr soon

MaEtUgR, 2017-03-01:
I posted the plot in a new issue because it is not in descending flight and therefore might potentionally not be exactly the same issue.

matanhavi, 2017-03-01:
@Stifael, Note that I haven't made change on the SITL params, you can reproduce it with the default parameters.
From user point of view, if you doing any flight mode, and then go to "pause" mode, you want the drone to stop moving as soon as possible, not to do any non-necessary overshoots. 
The question here is: are you prefer to set the SP at exact location where you set the mode? or to stop where ever you go to after the overshoot?
anyway the current logic isn't good.

Stifael, 2017-03-01:
@matanhavi:
The question is not where to set the setpoint, it is supposed to be set at the current location of the vehicle with some offset in the direction of flight. Can you also upload a log file from your flight? It would be helpful to analyze it better. Im also wondering what flight mode you used,.

matanhavi, 2017-03-01:
@Stifael, how can I upload ulg file?
anyway the first description is the full description of what I done, the flight modes are immidiate from takeoff, change alt to 100m, than to 50m, and than pause

Stifael, 2017-03-01:
https://dev.px4.io/flight_log_analysis.html.  I will try to reproduce it tomorrow.

matanhavi, 2017-03-02:
because its ULOG file, and not PX4log it doesn't allows me

Stifael, 2017-03-02:
ok, no worries. I quickly had time to test it with setting different altitude setpoints, and I can easily reproduce it. This issue is interesting since I found the slewrate issue during velocity control a couple days ago, where the change in velocity setpoint is also limited, which caused a delay in reaction time from large velocity to hold or to velocity with different sign. This issue here is indeed related, but it effects the outer controller (the position controller), and so far I did not even know that there is a slewrate applied to the position setpoint as well. I mean, this issue becomes even more extreme if you fly downwards, but then at some point you change direction to upwards, but the vehicle would continue to descend until the setpoint reached the current state. 
tomorrow I will look into it.

Stifael, 2017-03-02:
by the way, you can upload ulog files to flight review. It is also described in the link I gave you.

matanhavi, 2017-03-03:
Thanks! its seems that you understood all my points. 
does  #6717 fixes also this issue?

Stifael, 2017-03-03:
It fixes the issue that you have. but I first need  to verify that this change does not effect other flight modes. I also work on changing the logic for mc in auto mode

Stifael, 2017-03-22:
@matanhavi  it should now be fixed at least to the extend that it does not have a delay in position setpoint when setting a target position

closed 2017-03-22 by Stifael