PX4/PX4-Autopilot #6685: enabling mpu9250 and hmc5883 on navio2

nicolaerosia, 2017-02-27:
Is there a reason why mpu9250 and hmc5583 are not being used on Navio2 ?
https://github.com/PX4/Firmware/blob/master/posix-configs/rpi/px4.config#L9-L10
Could someone please shed some light on this?

Thank you

LorenzMeier, 2017-02-27:
Probably merely an artefact of the testing setup. You should be able to enable them instead. I would need to check if the wrappers are set up correctly for multi-sensors on Navio - you might want to enable only one set by default to start.

closed 2017-02-27 by LorenzMeier

nicolaerosia, 2017-02-28:
@mhkabir I believe you have Navio2. Could you please comment on this? Thanks!

mhkabir, 2017-02-28:
I don't have anything setup right now.

Why don't you test it and let us know? :)