PX4/PX4-Autopilot #6697: Unable to find current revision in submodule path 'src/lib/ecl'

potato77, 2017-03-01:
Hello,

I used the cmd git submodule update --init --recursive to update all submodules of px4 .

and the error is:

fatal: Needed a single revision
Unable to find current revision in submodule path 'src/lib/ecl'

How to fix it? Thank you in advance!

any help would be appreciated!

mhkabir, 2017-03-01:
Please run `make submodulesclean`

potato77, 2017-03-01:
$ make submodulesclean

Usage: git submodule [--quiet] add [-b branch] [-f|--force] [--reference <repository>] [--] <repository> [<path>]          or: git submodule [--quiet] status [--cached] [--recursive] [--] [<path>...]                                            or: git submodule [--quiet] init [--] [<path>...]                                                                       or: git submodule [--quiet] update [--init] [-N|--no-fetch] [-f|--force] [--rebase] [--reference <repository>] [--merge] [--recursive] [--] [<path>...]                                                                                         or: git submodule [--quiet] summary [--cached|--files] [--summary-limit <n>] [commit] [--] [<path>...]                  or: git submodule [--quiet] foreach [--recursive] <command>                                                             or: git submodule [--quiet] sync [--] [<path>...]                                                                    make: *** [submodulesclean] Error 1

mhkabir, 2017-03-01:
That's pretty odd. What operating system and version are you using? 

@dagar any suggestions?

potato77, 2017-03-01:
windows 10

dagar, 2017-03-01:
@potato77 have you successfully built PX4 on windows before? What branch are you on?

potato77, 2017-03-01:
@dagar no. This is my first time to install the development environment.

dagar, 2017-03-01:
Can you walk us through what you've done so far, which tools you've installed. It's not clear to me if you have a really old version of git or there's some Windows path or line ending issue.

If you're able to use Linux (Ubuntu) or OS X for development you'll have a much easier time. It's possible to get this working in Windows, but much slower and poorly supported.

potato77, 2017-03-01:
@dagar thank u very much
I am trying to install the development environment on the VM unbuntu.

mhkabir, 2017-03-01:
I believe we could eventually create a setup that works on Bash for Windows 10, thus avoiding all the issues we're currently having. 

I had a quick look-see and it seems like it'll be easy to do.

dagar, 2017-03-01:
That would certainly be nice. If there's anything to do to get it working we can then keep it working by adding a build on https://www.appveyor.com/.

potato77, 2017-03-01:
@mhkabir any instruction to Bash for WIN10? How to build PX4?

mhkabir, 2017-03-01:
@potato No instructions yet. I don't have a Windows 10 machine running at the moment. I can however assist you in bringing it up remotely, later today. Skype me : 'kabiriris'.

mhkabir, 2017-03-01:
Start by installing Bash if you don't have it already : https://www.howtogeek.com/249966/how-to-install-and-use-the-linux-bash-shell-on-windows-10/

mhkabir, 2017-03-01:
@dagar I don't think Appveyor has support for Bash on Ubuntu yet : https://github.com/appveyor/ci/issues/1295

mhkabir, 2017-03-05:
If anyone running Windows would like to help me with testing, it would be awesome. You can email me or ping me on Skype, and I'll guide you through the process. 
We can then update Devguide with a working set of instructions for everyone to use.

dagar, 2018-01-17:
This works for me.