PX4/PX4-Autopilot #6698: Difference between "make px4fmu-v2_default" and the "make -j1 px4fmu-v2_default" in px4 toolchain

zimoqingfeng, 2017-03-01:
Hello,

I used the cmd 'make px4fmu-v2_default' and 'make -j1 px4fmu-v2_default'.
The OS is windows 10, version:px4_toolchain_installer_v14_win.exe

I failed to build the PX4 by the first command at the first time, 
and i tried the second command , then sucessfully built it,
maybe the second command is much  more faster than the first during the compiling,
and i tried the first command again, it became successful,

And i would like to know the difference between the two commands,

Thank you in advance!
any help would be appreciated!

BR,

dagar, 2017-03-04:
The actual build system is cmake, but we have a Makefile wrapper to try and keep it simple. You shouldn't specify parallel flags at the top level.

My guess is on Windows the first time the build failed for an unrelated reason (race condition or bad build dependency) and the second time it succeeded. Windows is not well supported for PX4 development as most developers are using Ubuntu (natively or in Virtualbox/VMWare), or OSX.

There's some interest in improving the Windows dev experience here, https://github.com/PX4/Firmware/issues/6697

zimoqingfeng, 2017-03-05:
Thanks for your guide! And i will check it more clearly.

closed 2017-03-08 by julianoes