PX4/PX4-Autopilot #6699: how to use the shell file  sitl_multiple_run.sh to run a multiple drone in gazebo simulation

kiti1013, 2017-03-01:
when i run `sudo ./ sitl_multiple_run.sh `
in the (/Firmware/build_posix_sitl_default) produce instance_1 instance_2 floder  and  the out.log prompt me to  Please start the flight simulator to proceed..
but i don't know how to do in the next?
who can help me ?
thanks.

kiti1013, 2017-03-01:
@LorenzMeier @jdzerve @korotkoves

korotkoves, 2017-03-01:
You should connect multiple drones in Gazebo (with sitl_gzebo) to px4 instances started by sitl_multiple_run.sh. See/use https://github.com/acsl-mipt/multiple-sitl for example.

kiti1013, 2017-03-02:
thank   you  @korotkoves  , i haved run  your code in ubuntu16.04 and everything is ok , i am trying to run this code in ubuntu 14.04.

closed 2017-04-11 by dagar

paan2097, 2017-04-21:
I want to do multiple planes in SITL using PX4 using this file:
https://github.com/PX4/Firmware/blob/master/Tools/sitl_multiple_run.sh
Two instances of PX4 are started but reach this line:
Please start the flight simulator to proceed..
I want to connect to these two px4 sitl instances using dronekit scripts....but it is telling me I need to start a gazebo...How do I create either two gazebos, or even better one gazebo with both planes?
The https://github.com/acsl-mipt/multiple-sitl repo isn't setup for planes
Thanks!