PX4/PX4-Autopilot #6702: Horizontal velocity control input delay

MaEtUgR, 2017-03-01:
If you have the cruise velocity high, the maximum tilt angle pretty low and the slew rate (maximal horizontal acceleration) on a reasonably low value you experience a pretty noticable delay between command and vehicle reaction. This problem was also reported in atomatic flight modes.

Digging into it @Stifael and I found out that the velocity controller looses close tracking because of the tilt angle limitation while the setpoint goes to your high valued cruise velocity. Now if you want to stop or fly the oposite direction again, the slew rate prevents the fast setpoint change and the vehicle shows no reaction at all at least for the time the velocity setpoint is still higher than the actual velocity.

Here is a plot showing this effect with high cruise velocity of > 10m/s, a very small maximum tilt angle of 10 degree to clearly show the problem, and a reasonable maximum acceleration of the same value as the crusie velocity > 10 m/s^2.
[image]
turquoise stick input, yellow velocity setpoint, red attitude setpoint

It looks to me that this problem is closely related to https://github.com/PX4/Firmware/issues/6681

Stifael, 2017-03-01:
The main fundamental problem with the slewrate is that it is feedfoward-like implemented, independent of the current state. this means that slewrate limit only makes sense if it can be assured that the velocity tracking is guaranteed. 
A simple example illustrates the issue:
Flying against wind, where the local velocity of the vehicle is exactly zero but the velocity setpoint around 10m/s. If now the user demands opposite velocity and the wind calms down, the vehicle will accelerate in the same direction before the demand switch until the velocity setpoint reaches the current velocity.

LorenzMeier, 2017-03-01:
The fix to that would be to constrain the slew rate both based on the current reference and the current state, so that the reference cannot depart from the current state at more than the slew rate limit.

Stifael, 2017-03-01:
@Lorenz. 
Im not sure if that works. the septoint of the slewrate is as follow: vel_sp = max_acc * dt + vel_sp_prev, where acc_max * dt can be a rather small number. if you now do something like vel_sp = acc_max * dt + vel_current, then the vehicle will not move at all if the (sp-current_state) error is not large enough. In general, there is nothing bad with having a step input.
 If the goal is to reduce acceleration, then I suggest to do it after the velocity controller, since the output of the velocity controller is similar to acceleration

LorenzMeier, 2017-04-12:
Fixed / wip.

closed 2017-04-12 by LorenzMeier