PX4/PX4-Autopilot #6703: Camera trigger does not work on Pixhawk board

jdzerve, 2017-03-01:
Created thread in forum:
http://discuss.px4.io/t/camera-trigger-not-working-on-pixhawk/2610
But this looks more like a bug in FW version 1.5.5.
Configured PX4 on Pixhawk to do Time based trigger always on with:
TRIG_ACT_TIME: 500ms
TRIG_INTERVAL: 1000ms
TRIG_PINS: 123456
Expected to have equal PWM outputs on all 6 AUX ports.
This is what I get:
With RC transmitter ON:
AUX1: constant 1500 regardless of what is configured in Radio setup.
AUX2: constant 1500 regardless of what is configured in Radio setup.
AUX3: constant 1500 regardless of what is configured in Radio setup.
AUX4: constant 1500 regardless of what is configured in Radio setup.
AUX5: No signal
AUX6: No signal
With RC transmitter OFF:
AUX1: 1500 for 0.5s, 1700 for 0.5s (expected behavior)
AUX2: constant 1500
AUX3: 1500 for 0.5s, 1700 for 0.5s (expected behavior)
AUX4: constant 1500
AUX5: No signal
AUX6: No signal

Turns out that I can get camera trigger working only while RC is turned OFF and only on two pins which does not make sense.

mhkabir, 2017-03-01:
Looks like the AUX mixer is messing with it.

davids5, 2017-03-01:
I think there is a resource conflict on the timer channels because the camera trigger code assume it owns the timer when it does not. The FMU owns the timer so the fmu must be told to not user the channels the camera trigger code uses. The camera trigger code would benefit from a overhaul.

mhkabir, 2017-03-01:
@davids5 I believe we shouldn't be directly accessing the hardware like we currently are. Do you have the chance to do a refactor?

kd0aij, 2017-03-01:
> Is this Pixhawk specific or broken on other HW, too?

It would be STM32 specific

davids5, 2017-03-01:
@mhkabir - Yes- you are absolutely correct.

I am working on adding support to Kinetis K66 SoC. In that effort have to cleanse the tree of all STM32 code that is out of place and using resources directly. That code has to be in the drivers/<arch:<STM32:kinetis:>> directory and use the io_timer api to co-exist; 

It will not be a quick process - there is a lot to do in upstream Nuttx and [on this branch](https://github.com/PX4/Firmware/pull/6438)

davids5, 2017-03-01:
@jdzerve - I think you should be able to determine the conflict and ensure the FMU is in the correct mode.

dagar, 2017-03-01:
Is the issue setting the FMU mode?
If so is there a way the camera trigger could properly change this after start, or do we need to coordinate FMU mode and camera_trigger startup in the init script?

davids5, 2017-03-01:
@dagar - I would think so but it needs to be looked at - my (poor) recollection is the camera trigger is hardcoded (compile time) to use a given timer and channels.  

>FMU mode and camera_trigger startup in the init script?

You would have to look at what configures the hardcoded (compile time) in the camera trigger on a board basis. I hope there are not many of them. 

The correct solution it to use/add/augment the io_timer driver code and use it's enforcement of channel  mode and allocation. But as it stands the FMU forces the mode it wants (deallocates and reallocates) at will. Hence calling the "Owner"

mhkabir, 2017-04-03:
Closing in favour of #6961.

closed 2017-04-03 by mhkabir