PX4/PX4-Autopilot #6707: tar -jxf gcc-arm-none-eabi-4_9-2015q3-20150921-linux.tar.bz2

potato77, 2017-03-02:
I am installing the toolchain.
I used the command tar -jxf gcc-arm-none-eabi-4_9-2015q3-20150921-linux.tar.bz2.
Then,
[image]
why i cant install the gcc-arm-none-eabi-4_9-2015q3-20150921-linux.tar.bz2

potato77, 2017-03-02:
can i use the command     **sudo apt install gcc-arm-none-eabi**  to install it?

kiti1013, 2017-03-02:
`wget https://launchpad.net/gcc-arm-embedded/4.9/4.9-2015-q3-update/+download/gcc-arm-none-eabi-4_9-2015q3-20150921-linux.tar.bz2`

davids5, 2017-03-02:
We are using 
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 5.4.1 20160609 (release) [ARM/embedded-5-branch revision 237715]

```wget https://launchpad.net/gcc-arm-embedded/5.0/5-2016-q2-update/+download/gcc-arm-none-eabi-5_4-2016q2-20160622-linux.tar.bz2```

closed 2017-03-02 by davids5