PX4/PX4-Autopilot #6708: EFK2 Airspeed sensor parameter name

AlexisTM, 2017-03-02:
Dear, 

There is some inconstancy  between the airspeed parameter names. It has 3 names : `ASP`, `ARSP` and `EAS`. 

* EKF2_ASP_DELAY - Airspeed delay
* EKF2_ARSP_THR - Airspeed fusion threshold
* EKF2_EAS_NOISE - Airspeed noise

dagar, 2017-03-02:
I'm happy to do the rename for consistency, but then users silently lose their configuration when upgrading (if they've touched these).
@priseborough @LorenzMeier Thoughts?

priseborough, 2017-03-03:
I would avoid changing parameter names now for the reason @dagar gives. The exception might be if the parameter name is misleading, which is not the case for these examples. What is more important is that the description for each parameter is clear and complete.

dagar, 2017-04-11:
I think we'll have to close this and try to name things more consistently going forward.

closed 2017-04-11 by dagar