PX4/PX4-Autopilot #6709: VTOL Standard FW_PSP_OFFSET

dagar, 2017-03-02:
VTOL standard transitions should respect FW_PSP_OFFSET.

Front transition.
 - ramp up so that vehicle is at FW_PSP_OFFSET by the end of transition (verify clean init for TECS pitch)

Back transition
 - hold FW_PSP_OFFSET
 - as a bonus and crude holdover until we have full position control we could add an additional pitch param for simple braking

FW actuators can be given full weight at the beginning of front transition, and possibly until back transition completes.

tubeme, 2017-03-03:
I have a huge study about the FW_PSP_OFF. We have a PSP offset in our frame about 4-5 degrees + for the FW. So we've had a three weeks of flying last year and tried all possible setups to make it work. We tried zillion settings for the PSP offset without any success and a lot of crashes and unsuccessful transitions etc. So at the end we just removed the PSP offset and created a hardware reflex in the wing such that the transition is successful all the time. Once in the FW the autopilot corrected the reflex and the flying was normal. We fly this way up to now. If you need the data it will take me probably couple of days to dig back all the logs and give you the more emblematic flights. Else we can test the development of this aspect of the transition. This does not function correctly for sure!

sanderux, 2017-03-05:
Hi @tubeme 
Nice to hear from you again.

Based on your comments i think the param you would want to tune is VT_FW_PITCH_TROM and not FW_PSP_OFF. Pitch trim is a constant that will survive during transitions, pitch setpoint offset (psp_off) does not, it only sets changes the setpoint outside of transitions.

We also started with mechanical offset but now we fly with PITCH_TRIM at 0.1 and that's probably too high. If you still have issues you might want to check your CG.

tubeme, 2017-03-06:
Hi @sanderux, I'm getting on track now.

@LorenzMeier Hmmmm strange logic! Regarding the physics of the aerodynamics it is not logical to be this way. But we find it out the hard way :). I will explain why it is not logical. After the transition is past in altitude and position the PSP Offset does not have any effect on the flying characteristics. It is applicable only to the Manual modes. And this is not a good design practice in my opinion because the pitch offset is a fundamental characteristic of the frame! It should be applicable all the time to all modes. All other tricks (hardware and software) like the reflex you and we are using in the transition are corner cases... Actually PSP offset is fundamental mostly to the VTOL, because I don't see any need for this parameter in FW and MC mode on their own.

dagar, 2017-03-29:
Implemented

closed 2017-03-29 by dagar