PX4/PX4-Autopilot #6711: HIL with jMAVSim not working

gamoreno, 2017-03-02:
I'm trying to use a Pixhawk to do HIL with jMAVSim following the instructions in this page, but it doesn't work.
[https://pixhawk.org/dev/hil/jmavsim](https://pixhawk.org/dev/hil/jmavsim)

I connect the Pixhawk to a USB port in my computer, and wait until the main LED is breathing blue. When I start jMAVSim, at some point the LED blinks green twice, then back to blue, and then starts blinking purple. At this point, the FMU B/E LED is blinking very fast.

In [this page](https://uav-lab.org/2016/08/31/px4-research-log-8-hardware-in-the-loop-hitl-simulation-using-jmavsim-on-ubuntu-14-04/), I found that others are having the same problem.

I found that the last stable version with which this works is 1.4.4, even with the latest version of jMAVSim. With that version, after I start jMAVSim, the LED blinks green, and then solid green when armed.

julianoes, 2017-03-03:
Thanks for reporting. I will look at this next week sometime.

julianoes, 2017-04-18:
This should work again: https://dev.px4.io/en/simulation-hitl.html#using-jmavsim-quadrotor.

closed 2017-04-18 by julianoes