PX4/PX4-Autopilot #6712: Pixhawk Pro lockup after mag cal

tumbili, 2017-03-03:
I just flashed latest master on a pixhawk v3 pro and noticed the following behavior:
1) Selected airframe -> reboot
2) Calibrated gyro, calibrated accel
3) Did the mag calibration (internal only) and got a message like "Mag 1 fail: TOUT"
Shortly after the board seemed to lock up (LED was solid red) and QGC lost connection.
When I rebooted the board the sensors were still calibrated but the airframe parameter was set to 0.
The second time I tried it everything was ok.

sanderux, 2017-04-18:
Can't reproduce on my pixhawk pro

sanderux, 2017-04-18:
Manually changing the CAL+MAG_XOFF did indeed crash the FC

sanderux, 2017-04-18:
@klopezal could you have a look

LorenzMeier, 2017-04-30:
@klopezal Can you re-try this on a current board?

klopezal, 2017-05-01:
Can't reproduce with V1.3 modifications, but I'd like to instrument to make sure it is not a hidden problem.

LorenzMeier, 2017-05-01:
Since the HW is not widely deployed and its unclear if this still persists moving out to next release.

klopezal, 2017-05-18:
Unable to reproduce on V1.2 with jumper, waiting for V1.3 to validate and close.

closed 2017-08-20 by LorenzMeier