PX4/PX4-Autopilot #6721: PX4 and ping command

shening, 2017-03-03:
Hi I was wondering if the Mavlink ping command was implemented on the PX4 in order to measure GCS to aircraft connection latency 

PING ( #4 )

A ping message either requesting or responding to a ping. This allows to measure the system latencies, including serial port, radio modem and UDP connections.

Thank you,

Sebastian

dagar, 2017-03-04:
It's not implemented in PX4, but it shouldn't be that hard to add. If you're interested in adding this I'd be happy to point you in the right direction.

http://mavlink.org/messages/common#PING

You would add a simple function in the PX4 mavlink module here - https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_receiver.cpp#L704

``` C
void
MavlinkReceiver::handle_message_ping(mavlink_message_t *msg)
{
    // send ping reply
}

```

mhkabir, 2017-03-05:
We should then expose the ping statics per-link via `mavlink status`.

I've been meaning to do this for a while, but if @shening is interested in doing it, that's awesome!

dagar, 2017-04-28:
Would love to have this is you're interested in implementing. Pull requests welcome!

closed 2017-04-28 by dagar

weiminshen99, 2018-03-28:
Hi, All,

Could you please tell me if this has been implemented yet? I just stated flying multiple UAVs and it would be great to have this handy. Thanks in advance. 

Wingman

weiminshen99, 2018-04-03:
Hi, @dagar,
I am interested in implementing this because I need it badly for my multi-PX4 swarms. Please let me know how.  Do I need to open a new issue on this? Thanks!

dagar, 2018-04-03:
@weiminshen99 it looks partially implemented. https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_receiver.cpp#L1718-L1728

Yes please open a new issue to discuss.