PX4/PX4-Autopilot #6722: QGC and flightgear with pixhawk HIL can not work.

goodman110110110, 2017-03-04:
I want to do the HIL fix-wing simulation with QGC 3.1.3, pixhawk(pix4, latest code) and flightgear 3.4.
But I can not find the simulation airframe for flightgear but the HILstar (xplane), so I choose the HILstar (xplane) and apply.
When start the simulation, the QGC ui can receive the data from the flightgear and the RC, but flightgear can not receive the data from QGC, the plane in the flightgear can not be controlled by RC.

Please help me. Thank you!

pgorczak, 2017-03-09:
**Bump** I am stuck on the same issue

bkueng, 2017-03-10:
There was a recent fix for HIL merged to master (https://github.com/PX4/Firmware/pull/6718). Can you try again?
If there is a communication issue between QGC and flightgear, it should be raised in QGC: https://github.com/mavlink/qgroundcontrol

pgorczak, 2017-03-13:
I built and flashed the firmware from master. Starting FlightGear from the HIL widget in QGC triggers errors à la

> EMERGENCY: Accel 0 fail: TOUT!

Plus, as @goodman110110110 mentioned, QGC receives vehicle data but e.g. increasing throttle has no effect in FlightGear.

Is this an issue with the airframe? Like @goodman110110110 , I've picked the HILstar (xplane) one.

goodman110110110, 2017-03-14:
yes, I have the same problem.
The pixhawk firmware will never support the flightgear HIL?

Why current master has not the flightgear hil airframe but xplane.

The net should give the retail operator of how to use pixhawk hil with qgc
and flightgear in windows.

what can I do to carray out the HIL with flightgear.

lupusorina, 2017-04-26:
Hey, 
I am having the same issue. I am trying to do HIL with last version of PX4, QGC and Flightgear. It would be awesome if anyone found a solution/fix.

Thanks

closed 2018-02-21 by PX4BuildBot