PX4/PX4-Autopilot #6723: multicopter flies randomly when given small velocity setpoint in SITL offboard control

gaochangyu, 2017-03-05:
This problem occurs to me in version v1.5.5. I have a simulation program which uses mavros to publish velocity setpoint to control the quadcopter in SITL. The program works well in ROS indigo with Firmware v1.4.4. However, with Firmware v1.5.5 and ROS kinetic, the vehicle flies randomly and even turns over when I give a very small velocity setpoint.
I think it's not my program's problem. Because directly runing the gazebo simulation and using mavros to send vx,vy,vz = (0,0,0) setpoint to px4 also causes the  problem.
The behavior of the vehicle looks like the velocity control loop fails so the velocity oscillates and finally goes out of control.
I'm very confused. Any help?

gaochangyu, 2017-03-05:
Reproduce the problem:
1. run `make posix_sitl_default gazebo` in Firmware source dir (ver v1.5.5).
2. run mavros and connect to the firmware with udp.
3. use rqt or rostopic to publish /mavros/setpoint_velocity/cmd_vel continuously, all setpoint are 0.
4. takeoff the drone and switch to offboard mode.

mhkabir, 2017-03-05:
@Stifael @MaEtUgR Can you please have a look.

gaochangyu, 2017-03-11:
@Stifael @MaEtUgR  can you please have a look?

Stifael, 2017-03-11:
Yep will have a look on monday.

gaochangyu, 2017-03-16:
Hi, @Stifael , do you get any result?

Stifael, 2017-03-16:
I am sorry, I need to postpone it to the weekend

Stifael, 2017-03-20:
@gaochangyu, I finally was able to test it quickly.
I used ros kinetic and the newest master of px4 firmware. here is the log:
http://logs.px4.io/plot_app?log=acf5409a-5b7e-4a4c-805a-bb4cdb87333c

I did not encounter any difficulty with send the velocity setpoint over mavros. Can you also upload a log file?

dagar, 2017-04-12:
@gaochangyu please reopen with a log file if this is still an issue.

closed 2017-04-12 by dagar