PX4/PX4-Autopilot #6724: Accel & Gyro Fail while running simulation

tonyvanhain, 2017-03-05:
I am currently trying to run the offb_node example from the https://dev.px4.io/ website using gazebo and ros, and the simulations works, however I am getting error messages that looks like this:
ERROR [sensors] Accel #0 fail:  TOUT!
ERROR [sensors] Gyro #0 fail:  TOUT!

I got the messages continuously as I run the simulation. Is there any way to fix this, or is it because I am not connected to any hardware at all? also may I blog about how to run the offb_node example? I want to explain it more step by step on how to do it, because it will be very helpful for beginners like me. I will help with posting it to the dev.px4.io website if it is necessary.

mhkabir, 2017-03-05:
I've seen this happening on setups where the computer is too slow to run the simulation. Do you see any messages on the console indicating that?

tonyvanhain, 2017-03-05:
@mhkabir I do not see any signs indicating that on my console, but I do notice that my computer ran slower when I ran the simulation, can that be the indicator as well?

closed 2017-03-10 by tonyvanhain

alexcherpi, 2017-03-14:
Hi, I have the same issue than tonyvanhaim, I am running the simulator on Virtual Box.

rudja, 2017-03-16:
Hello @tonyvanhain. I am a french student and I am also working on this offb_node example using gazebo and ros. How did you succeed in using it in gazebo pls? For the moment, I have a ros and gazebo environment working but I don't know how to simulate this program in Gazebo...

smnogar, 2017-03-28:
Just to add to this thread, I encountered these errors when trying to run MAVROS and PX4 on a Snapdragon Flight. The issue is that I had set the mavlink rates too high for the ROS node to keep up.