PX4/PX4-Autopilot #6729: How to listen Multi publish uOrb topic.

winszenYoung, 2017-03-06:
I read Doc about reading uorb topic( using this doc : https://dev.px4.io/advanced-uorb.html  )
It works very well,
but, I have some problem
I want to read  multi publish topic.

For example. 
ls /obj      
->result :   ……
                  sensor_accel0                                                                  
                  sensor_accel1  
                  ……

nsh > listenr sensor_accel0                     -> it occur error
nsh > listenr sensor_accel1                     -> it occur error

but
nsh > listenr sensor_accel                     -> it works

sensor_accel is multi pusblished same name.
So, I don't read seperate data.

How do I get each uorb topic?

mhkabir, 2017-03-06:
I don't think listener works for multi topics. @tumbili Can you confirm?

bkueng, 2017-03-06:
@mhkabir you're right, it can only show the first instance.

@winszenYoung we welcome pull requests if you want to implement it.

closed 2017-03-06 by bkueng