PX4/PX4-Autopilot #6730: Land detected after VTOL back transition

sanderux, 2017-03-06:
Under certain conditions a false positive can occur on land detection with a VTOL.
http://logs.uaventure.com/view/t293nDhydn76rU94mnoZRL

ref: https://github.com/PX4/Firmware/pull/6728

sanderux, 2017-03-06:
@MaEtUgR i've analyzed further and it has been a strange, but not uncommon turn of events.
The conditions for the MC land detector where met mid air (https://github.com/PX4/Firmware/blob/master/src/modules/land_detector/MulticopterLandDetector.cpp#L251-L265)

[image]

It seems to me we need to verify the persistence of these conditions over a time period. What do you think?

sanderux, 2017-03-06:
ping @AndreasAntener

tubeme, 2017-03-07:
Is this applicable to 1.6.0 RC1??

sanderux, 2017-03-07:
@tubeme yes, i recommend you lower LNDMC_XY_VEL_MAX to something like 0.5
I will look in to making this procedure safer by adding time persistance

tubeme, 2017-03-08:
Thanks @sanderux  We will change it to 0.5

dagar, 2017-06-27:
Has this been resolved?

sanderux, 2017-06-27:
No, these are still constants:
https://github.com/PX4/Firmware/blob/d50a0c43239488fa472b96618dfe582017ba1b50/src/modules/land_detector/LandDetector.h#L143-L149

MaEtUgR, 2017-07-05:
@sanderux I read your detailed problem description here again: https://github.com/PX4/Firmware/pull/6735#issuecomment-285650891

**Assumptions:** I noticed that you are mentioning that the VTOL is in this moment an MC which is suposedly position controlled and in an auto landing state.

I verified using your log from the starting post of this thread that you were flying a position controlled mission, in this mission there was a sudden land detection and then you took over in manual mode to save your vehicle:
[image]

**Probable root cause:** @Stifael just found out that there is a duplicate "land/ground contact detection" mechanism inside the mc_pos_controller which is only enabled during AUTO_LAND. I would say that if you only had that problem in AUTO_LAND so far it is very likely that this is the root cause of your issue because the duplicate mechanism is very permissive for false positives and your description matches well.
Sadly I can not be sure because the required internal states do not get logged.

**Probable solution:** @Stifael recently worked on removing this premissive duplicate ground detection and adopting only the general idea not dependet on AUTO_LAND mode to the first stage of the real land detector. Which means better architecture, no loss of functionality and if there is ever a false positive it only enters the first stage and will immediately keep on flying.

Feel free to test the pending PR, always happy about feedback: https://github.com/PX4/Firmware/pull/7314
As a reference, here's the duplicate removal part: https://github.com/PX4/Firmware/pull/7314/files#diff-fc77c3ef569029d45764664c75d4b0c1L1868

closed 2017-10-18 by sanderux