PX4/PX4-Autopilot #6738: building the firmware on cygwin

b-rohit, 2017-03-07:
hi, I am new to drones. I am trying to build the firmware on cygwin platform. I got following error >
[Rohit:/cygdrive/e/Dw/px4f/Firmware]$ make px4fmu-v2_lpe
CMake Error: Could not create named generator MSYS Makefiles

Generators
  Unix Makefiles               = Generates standard UNIX makefiles.
  Ninja                        = Generates build.ninja files.
  CodeBlocks - Ninja           = Generates CodeBlocks project files.
  CodeBlocks - Unix Makefiles  = Generates CodeBlocks project files.
  CodeLite - Ninja             = Generates CodeLite project files.
  CodeLite - Unix Makefiles    = Generates CodeLite project files.
  Eclipse CDT4 - Ninja         = Generates Eclipse CDT 4.0 project files.
  Eclipse CDT4 - Unix Makefiles= Generates Eclipse CDT 4.0 project files.
  KDevelop3                    = Generates KDevelop 3 project files.
  KDevelop3 - Unix Makefiles   = Generates KDevelop 3 project files.
  Kate - Ninja                 = Generates Kate project files.
  Kate - Unix Makefiles        = Generates Kate project files.
  Sublime Text 2 - Ninja       = Generates Sublime Text 2 project files.
  Sublime Text 2 - Unix Makefiles
                               = Generates Sublime Text 2 project files.

/bin/sh: line 0: cd: /cygdrive/e/Dw/px4f/Firmware/build_px4fmu-v2_lpe: No such file or directory
make: *** [Makefile:145: px4fmu-v2_lpe] Error 1

mhkabir, 2017-03-07:
Please use Ubuntu in a VM or natively to make life easier. Windows support isn't being actively maintained. 

There is also an issue tracking Bash on Windows 10 support, which we prefer. If you want to help test it, please let me know.

b-rohit, 2017-03-07:
Okay thanks . I will build in VM

closed 2017-03-07 by b-rohit