PX4/PX4-Autopilot #6749: Snapdragon Flight IMU offset and calibration

ChristophTobler, 2017-03-08:
@jywilson I currently have a weird IMU offset (see image) and I'm not able to calibrate the Accelometer using QGC. Is it not possible via QGC or maybe there's another way?
[image]

Here's the output from PX4:
```
px4 starting.

INFO  [sdlog2] [blackbox] /root/log/2017-03-08
INFO  [sdlog2] [blackbox] recording: 07_13_42.px4log
INFO  [dataman] Unkown restart, data manager file '/home/linaro/dataman' size is 11797680 bytes
error finding param: VT_B_TRANS_DUR
error finding param: FW_CLMBOUT_DIFF
error finding param: VT_B_TRANS_DUR
error finding param: VT_B_TRANS_DUR
error finding param: VT_B_TRANS_DUR
error finding param: VT_B_TRANS_DUR
error finding param: VT_B_TRANS_DUR
error finding param: VT_B_TRANS_DUR
error finding param: VT_B_TRANS_DUR
error finding param: VT_B_TRANS_DUR
error finding param: VT_B_TRANS_DUR
error finding param: FW_THR_CRUISE
INFO  [mavlink] mode: Normal, data rate: 1000000 B/s on udp port 14556 remote port 14550
Sleeping for 1 s; (1000000 us).
pxh> INFO  [mavlink] using network interface wlan0, IP: 10.20.21.87
INFO  [mavlink] with broadcast IP: 10.20.31.255
INFO  [mavlink] partner IP: 10.20.21.14
```

and from the mini-dm: http://pasted.co/028df77a

ChristophTobler, 2017-03-08:
I tried another board and there the IMU seems to be normal. However, I still get the same PX4 and mini-dm output with the errors and sensor calibration does not work with QGC.

Maybe the other IMU is just broken?

ChristophTobler, 2017-03-08:
I also tried calibration with a GPS attached. Nothing happens when I press on calibrate

julianoes, 2017-03-08:
I have seen IMUs that are off like this and we usually rejected it as "bad hardware".

The calibration issue is separate and it seems it got broken recently.

ChristophTobler, 2017-03-08:
https://github.com/PX4/Firmware/pull/6753 solves the calibration issue and the IMU issue is due to bad hardware I guess...

closed 2017-03-08 by ChristophTobler

jywilson, 2017-03-08:
@ChristophTobler 
@julianoes is right about possible hardware issues with certain boards.  I would try a different board as Julian suggests. If this makes a difference, please get an RMA from Intrinsyc and return the board to them for a new one.