PX4/PX4-Autopilot #6752: QGC showing wrong battery percentage when throttle is increased

lakshaydang, 2017-03-08:
Hey everyone i am having some issues regarding battery calibration while working with pixhawk. The battery percentage which i am getting on my QGC is right when it is not armed but when i armed the quad and increase its throttle then instead of decreasing the battery percentage it started increasing. For example if my battery is 74 percent remaining, now when i increase its throttle it started increasing to approx 88% at full throttle instead of decreasing.
I am badly stuck up unable to understand and solve the problem. Please help.

nanthony21, 2017-03-30:
As you increase the throttle the voltage of the battery drops. Normally this would cause it to appear that % remaining battery was lower than it actually is. PX4 counters this by scaling the voltage based on the current throttle. If you look in the advanced battery settings in QGroundcontrol there will be a box where you can the voltage drop caused by 100% throttle. It sounds like the value of this parameter may be too high for you. Try reducing it until your % battery remaining is unaffected by changes in throttle.

lakshaydang, 2017-04-01:
ok i will try it and let you know about it.
Thanks

dagar, 2017-04-11:
@lakshaydang please reopen if still an issue.

closed 2017-04-11 by dagar