PX4/PX4-Autopilot #6754: Firmware not running in STM32F4DISCOVERY

dcabecinhas, 2017-03-08:
I can manage to bootload the firmware successfully but then it will only run for 5 seconds and crash the board (activity light blinks fast for 5s, then stops and the USB device disconnects).

My setup

- Board STM32F407VG Discovery
- Linux environment Ubuntu 16.04.2 LTS following instruction at https://dev.px4.io/starting-installing-linux.html 
- Toolchain gcc-arm-none-eabi-5_4-2016q2-20160622

Steps taken

- Connect board using the mini-USB cable
- Compile and upload PX4 Bootloader (see [bootloader.txt](https://github.com/PX4/Firmware/files/828145/bootloader.txt))
- Connect board using micro-USB cable
- Check serial port properties (see [serial_port_info.txt](https://github.com/PX4/Firmware/files/828144/serial_port_info.txt))
- Compile and upload PX4 Firmware (see
[firmware.txt](https://github.com/PX4/Firmware/files/828146/firmware.txt))
- Connect to /dev/ttyACM0 to get to Nuttx shell (**fails**). 

The serial port after the Firmware is uploaded last just around 5 seconds, does not output anything, and then the whole board crashes.

davids5, 2017-03-08:
@dcabecinhas see https://github.com/PX4/Firmware/issues/6613 about level of current support

LorenzMeier, 2017-03-11:
Closing as duplicate.

closed 2017-03-11 by LorenzMeier