PX4/PX4-Autopilot #6755: Analyzing rplidar data in gazebo or rqt_plot

nahush8, 2017-03-08:
With reference to this post : Add iris_rplidar model #6558 , jgoppert  I am running the appropriate launch file and the iris with rplidar is hovering properly in offboard mode. I want to analyise the lidar data and I am trying to check /laser/scan topic ? The intensities are not changing eve when i add an obstacle in between. Am I doing anything wrong ?

mhkabir, 2017-04-03:
The lidar model uses the Gazebo `GpuRay` class, which needs a GPU to execute. You can try switching it to the standard CPU one, which will work.

closed 2017-04-03 by mhkabir

lolster, 2018-03-22:
We are facing a similar issue using Gazebo 7 and ROS Kinetic. When using `GpuRay` for the `iris_rplidar` sitl model the ranges show `-inf` as values. 
When switching to `Ray` the simulation fails with a SegFault. We are using the PX4 v1.7.3.  
We are running the simulation through `roslaunch px4 mavros_posix_sitl.launch vehicle:=iris_rplidar`.

mhkabir, 2018-03-22:
Do you also have a laser altimeter on the model? Please open a new issue.

cozavci, 2018-03-22:
I had the same issue. I removed lidar altimeter from model and rplidar works fine now.

lolster, 2018-03-23:
@mhkabir Thanks for the quick reply, I will open an issue shortly.  
Just to be clear, to switch to `Ray` I just needed to change the `model.sdf` of rplidar and change 
sensor type from `gpu_ray` to `ray`, and plugins from `libGpuRayPlugin.so` to `libRayPlugin.so` and `libgazebo_ros_gpu_laser.so` to `libgazebo_ros_laser.so`.

@cozavci Thanks for the reply, I'll certainly try that.

BTW I just tried the GPU ray version of RPLidar again and it worked this time and I was able to get what I believe are consistent readings. I suspect it's to do with my graphics drivers.

Anyway, I shall open an issue soon about this, since the `Ray` version does not work for us.

sarahToonsi, 2018-12-19:
can you please explain what do you mean by laser altimeter cant find that in my rplidar SDF file. I also looked in the SDF file of the iris that Im using no laser altimeter

kashifzr, 2024-04-15:
@mhkabir, In sdf file inside Model folder/directory I did not see GPU base GPUray in PX4-1.14.0. Even then i find its hard to solve this issue. Even in build directory where libgazebo_ros_plugin.so is found i did not see any libgazebo_ros_gpu_plugin.so. 

Below is code of sdf file 
      <sensor name="laser" type="ray">
        <ray>
          <scan>
            <horizontal>
              <samples>360</samples>
              <resolution>1</resolution>
              <min_angle>-3.14</min_angle>
              <max_angle>3.14</max_angle>
            </horizontal>
          </scan>
          <range>
            <min>0.2</min>
            <max>6</max>
            <resolution>0.05</resolution>
          </range>
          <noise>
            <type>gaussian</type>
            <mean>0.0</mean>
            <stddev>0.01</stddev>
          </noise>
        </ray>
        <plugin name="laser" filename="libRayPlugin.so" />
        <plugin name="gazebo_ros_head_rplidar_controller" filename="libgazebo_ros_laser.so">
          <topicName>laser/scan</topicName>
          <frameName>rplidar_link</frameName>
        </plugin>
        <always_on>1</always_on>
        <update_rate>10</update_rate>
        <visualize>true</visualize>
      </sensor>
    </link>
  </model>
</sdf>