PX4/PX4-Autopilot #6760: Pixfalcon telemetry port rate can't be changed by SD card extras.txt file 2

0xd1ma, 2017-03-09:
hello

I`m trying to change baudrate on Pixfalcon TELE1 in accordance with
https://github.com/PX4/Firmware/issues/3543

On SD cart created:
/etc/extras.txt

Inside extras.txt:
mavlink stop-all
mavlink start -d /dev/ttyS1 -b 921600 -r 20000

then trying to connect with mavros:
roslaunch mavros px4.launch fcu_url:="serial:///dev/ttyACM0:921600"

but it does`t work.

please help with it

0xd1ma, 2017-03-15:
Please give a hint. where is in source files change TELE1 baudrate.

dkircali, 2017-03-20:
inside sd card I created etc/extras.txt with those lines:

#!nsh
mavlink stop-all
mavlink start -d /dev/ttyS1 -b 921600 -r 3000
mavlink start -d /dev/ttyACM0

I am able to communicate with 921600 baudrate. Also If you want higher imu publish rate later you can try adding this line:

mavlink stream -d /dev/ttyS5 -s HIGHRES_IMU -r 200

mhkabir, 2017-04-03:
Use `mavlink start -d /dev/ttyS1 -b 921600 -r 20000 -m onboard`

closed 2017-04-03 by mhkabir