PX4/PX4-Autopilot #6762: S.BUS scanner: Do only execute if HW inverter is present

LorenzMeier, 2017-03-09:
Due to the lack of a checksum on S.BUS it can be incorrectly detected. We need to disable the scanning for it if no HW support / inverter is present.

philippamlong, 2017-03-17:
I have a frsky l9r connected to pixracer with sbus. In master, no rc input is detected. In latest stable, it works. Maybe this is the reason?

LorenzMeier, 2017-03-17:
No, that must have a different reason. @davids5 Did we change the inverter control pin config?

davids5, 2017-03-17:
@LorenzMeier 

Any RC changes are still out on a [PR](https://github.com/PX4/Firmware/pull/6777) 

It was vetted by @pkocmoud with FrSky X4R- SB using sBus. 

How does the frsky l9r compare the the FrSky X4R- SB using sBus? 

@philipoe can you try the above PR and see if it solves your issue?

philippamlong, 2017-03-17:
I tried the PR, i did not work. Same behavior as in master. Also i tried self compiled master and travis master from qgroundcontrol.

What i did to reproduce:

Flash Master / PR 6777
Reset all params in qgroundcontrol
reboot
select airframe wing wing
reboot
radio -> calibrate
error "Detected 0 radio channels. To operate PX4, you need at least 5 channels"

When i do the same steps with stable or beta branch (selfcompiled and travis tested), it works as it should, all channels are working.

My pixracer is from banggood (clone?)

pkocmoud, 2017-03-18:
I can confirm that the sBus support in master is very  intermittent.  I can use the RC input thru calibrating the Radio portion of the Flight Controller bring up. After a reboot or IMU calibration routines the sBus receiver input stops working. I am only able to recover by running "mtd erase" and "param reset" and starting over. Loading beta or stable in master does not have this intermittent-icy. Not sure what to try next.

davids5, 2017-03-21:
@LorenzMeier 

@pkocmoud did some regression testing and determined the root cause to be the reset not restarting the FMU.  

1) With the system booted with the radio connected it fails.
2) With the system booted and then the radio connected it works.

We then tested reverting these commits and it worked.

63857c6afcc719a060c986a20410e8670c1aa971
a716b105f545188b79a3e552f23cad864d1c1821
b8e72789cd561512e4d13f8a5168ec2e89e732fd

I suspect the recreation of the FMU is is not initializing the sbus scanning.

vinhk, 2017-03-24:
I have similar issue after flashed masters using 'make px4fmu_v2_lpe. I got this error  "Detected 0 radio channels. To operate PX4, you need at least 5 channels" when calibrate Radio in QGC.

davids5, 2017-03-28:
There are 2 problems with the same symptoms:

Please see 

1) px4fmu-v4 build see https://github.com/PX4/Firmware/pull/6907

2) px4fmu-v2, px4fmu-v3 builds see https://github.com/PX4/Firmware/issues/6882#issuecomment-289615062

closed 2018-02-21 by PX4BuildBot