PX4/PX4-Autopilot #6765: Snapdragon

ChristophTobler, 2017-03-09:
@jywilson @julianoes @Stifael 
Quick update for you guys and maybe you know the source of some errors.
current standing:
* All the sensors seem to be working (including flow and range finder) on the different connectors
* Handheld tests show reasonable output

current issues:
* I cannot arm. I have calibrated all the sensors, the only thing that is still red in the QGC sidebar is battery, but I don't think that this has an effect since all preflight checks pass and the tab is battery tab is empty. If execute `arm` in QGC nothing happens. ~10 secs later QGC says `Vehicle did not respond to: arm/disarm`. Mini-dm outputs: `HAP:52:TEST px4muorb_topic_advertised of [vehicle_command] on remote side...  0167  px4muorb.cpp`. If I execute `qshell commander arm`, it does arm and detect a takeoff but no motors spinning.
* It can happen that the QGC alert overflows with the error `Critical: REJECT POSITION CONTROL` even though it is in `Manual` (also on the RC).
* Something minor: When aborting `px4 mainapp.config`, a lot of times it takes some time and ouputs
```
WARN  [uorb] [uORB::DeviceNode::publish(460)]: Error Sending [manual_control_setpoint] topic data over comm_channel
WARN  [uorb] [uORB::DeviceNode::publish(460)]: Error Sending [manual_control_setpoint] topic data over comm_channel
WARN  [uorb] [uORB::DeviceNode::publish(460)]: Error Sending [manual_control_setpoint] topic data over comm_channel
WARN  [uorb] [uORB::DeviceNode::publish(460)]: Error Sending [telemetry_status] topic data over comm_channel
WARN  [uorb] [uORB::DeviceNode::publish(460)]: Error Sending [manual_control_setpoint] topic data over comm_channel
WARN  [uorb] [uORB::DeviceNode::publish(460)]: Error Sending [manual_control_setpoint] topic data over comm_channel
WARN  [uorb] [uORB::DeviceNode::publish(460)]: Error Sending [manual_control_setpoint] topic data over comm_channel
WARN  [uorb] [uORB::DeviceNode::publish(460)]: Error Sending [manual_control_setpoint] topic data over comm_channel
```

Here's the mini-dm output: http://pasted.co/e7e0a48e

julianoes, 2017-03-09:
```
[08500/00]  13:02.005  HAP:37:Takeoff detected  2174  commander.cpp
[08500/00]  13:11.582  HAP:37:Landing detected  2172  commander.cpp
[08500/00]  13:16.436  HAP:63:TEST px4muorb_topic_advertised of [qshell_req] on remote side...  0167  px4muorb.cpp
// This is where it crashes on the DSP!
[08500/02]  13:38.870  HAP:110:cannot find /oemconfig.so  0141  load.c
```

jywilson, 2017-03-10:
This is a problem I have seen as well when testing from the master branch.  I am looking at it now.  Something must have changed that affects the state machine.  I can see that the arm.arm status changes when I move the joysticks to the lower right, but the motor driver does not receive the motor commands orb.

Jim W.

jywilson, 2017-03-10:
Okay, I have a patch to fix the arming problem.  I am not yet sure why it's needed, so this still WIP.  The PR is here: https://github.com/PX4/Firmware/pull/6767

Jim W.

ChristophTobler, 2017-03-10:
@jywilson Arming still doesn't work in QGC but with the RC it does. However the motors are still not spinning... @Stifael Any ideas why that is? I have increased `PWM_MIN` but didn't change anything.

Another thing that I realized: When I have e.g. `param set PWM_MIN 1300` in the `px4.config` it won't change the parameter. However it does work in `mainapp.config`. @julianoes is this normal?

julianoes, 2017-03-10:
> Another thing that I realized: When I have e.g. param set PWM_MIN 1300 in the px4.config it won't change the parameter. However it does work in mainapp.config. @julianoes is this normal?

I'm not sure, this should work both ways and everywhere in theory.

LorenzMeier, 2017-03-10:
@jywilson did the locking maybe only get removed on one side but should have gotten removed on both?

jywilson, 2017-03-10:
@ChristophTobler @julianoes @LorenzMeier  Regarding this question...

"Another thing that I realized: When I have e.g.  param set PWM_MIN 1300  in the  px4.config  it won't change the parameter. However it does work in  mainapp.config . @julianoes is this normal?"

Parameter setting in px4.config will only work if the qurt build defines the specified variable.  At build time two .XML files are automatically generated, one for the AppsProc and one for the aDSP.  If you want to change a parameter value in the qurt code, you need to be sure that the parameter is defined in a *params.cpp file that is compiled in the qurt build.  Try adding the parameter definition macro to a new or existing params file.

The inverse is true when trying to change a parameter from within AppsProc code.  The AppsProc code must be built with a *params.cpp file that includes the parameter definition.

So effectively parameters are *not* global between subsystems on Snapdragon Flight.  I am not sure if anything is done to detect duplicates, in case the parameter is inadvertently maintained in both places.  Making parameters global on a multi-subsystem SoC would require some enhancements to the current parameter infrastructure.

Jim W.

Jim W.

Jim W.

jywilson, 2017-03-10:
@LorenzMeier  
Regarding: "did the locking maybe only get removed on one side but should have gotten removed on both?"

This problem is related to something which happens when the parameters.xml file is generated.  The PARAMS_DEFINE for the MAV_TYPE parameter in the src/platforms/qurt/px4_layer/params.c file never get's copied into the XML file.  If I move the same PARAMS_DEFINE statement to a different params file, it does.  

I am still investigating this.

jywilson, 2017-03-10:
I have generated one additional PR for this problem.  It appears that if the comment in the PARAMS_DEFINE includes the "@board" tag, the parameter will not get copied into the parameters.xml.  

The PR is here:
https://github.com/PX4/Firmware/pull/6774

I will create an issue for the problem stated above.

Also, it appears that the parameter synching between the aDSP and AppsProc will not work if the same parameter is not defined for both builds.  This occurs because there are two parameters.xml files generated.  We are thinking that at one time a single parameters.xml was generated for both builds.  

I will create an issue for this as well.  

Jim W.

LorenzMeier, 2017-03-10:
@jywilson The board tag must match the board name, which it does not here. So that's not actually an issue. The fact that two parameter.xml files are generated sounds more like a real issue.

jywilson, 2017-03-11:
@LorenzMeier 
> The board tag must match the board name

Okay, I can make another change to add the correct board tag.  It would be helpful if the parameter generation tool output a statement of parameters that are being discarded.

I have confirmed the parameter.xml issue which I will create an issue for.  As it stands now, there is no way to synchronize parameters and if a parameter does not exist in both builds it can't be updated.  Since the update/synch process is broken putting the parameter in both builds won't work as a workaround, unfortunately.

ChristophTobler, 2017-03-13:
I also have some issues with RC: https://github.com/mavlink/qgroundcontrol/issues/4738
I cannot calibrate the RC because of that. Or has someone a solution to this?

mcharleb, 2017-03-13:
@jywilson "It would be helpful if the parameter generation tool output a statement of parameters that are being discarded." Sounds like an issue waiting to be filed :)

ChristophTobler, 2017-03-14:
Update:
* We found the reason for the QGC crash (but not the actual source... see mavlink/qgroundcontrol#4738)
* There seems to be an issue with the parameter sending/update. If I calibrate the RC, I don't think these parameters make it to the DSP side.
* The is again the `shem` error
[image]
[image]

jywilson, 2017-03-15:
Thanks for the update Christoph.  

The parameters sending/update is the result of the issue reported here: https://github.com/PX4/Firmware/issues/6778

I haven't seen the shmem error, but I think for now we should focus on the parameters.xml problem first.

jywilson, 2017-03-24:
Christoph,

This issue should be fixed with this PR:
https://github.com/PX4/Firmware/pull/6883

Jim W.

jywilson, 2017-03-28:
@ChristophTobler can you verify that this is mixed on the master branch, and close if so.  Thanks.

closed 2017-03-29 by ChristophTobler