PX4/PX4-Autopilot #6769: NOT ARMING: Preflight checks failed

bssung, 2017-03-10:
I use pixhawk with px4 firmware. I upload the firmware and setup using QGC. I have calibrated all the sensors,  and all the sidebar is green in the QGC. Then I connect the battery, and usb still connected , I try to arm using RC, the red light of pixhawk flash twice, and the QGC says "NOT ARMING: Preflight checks failed". 
I dont know which item is failed. How can I get detail information about the check failed? How to pass the check?

tubeme, 2017-03-11:
Either connect to the terminal with a terminal program like Putty. Or just get the SD card out read it with SD reader, find the pre-flight log in the card. Open it with TXT editor and see what is wrong.

tops4u, 2017-03-11:
Did you enter the Circuit Breaker Code for USB Connection otherwise PX4 will deny arming with active USB Connection. Also when first powered via USB and Battery connected later will usually cause a Problem with arming, usually you should power up properly from Battery only.

bssung, 2017-03-13:
@tops4u @tubeme  ,Thanks for your suggestion. I tried to power up from Battery only, there also an error "NOT ARMING: Connect power module." in the log of SD card. I roughly commented out this code below in arming_state_transition function in state_machine_helper.cpp, and upload the firmware, now I can arm my drone. 
But I dont kown why this power check is failed. Does this mean I set wrong value in the power setup in QGC? How can I pass the power check in a normal way? 
----------------------------------------------------------------------------------------------------------
if (!status_flags->circuit_breaker_engaged_power_check) {
// Fail transition if power is not good
		if (!status_flags->condition_power_input_valid) {
		mavlink_log_critical(mavlink_log_pub, "NOT ARMING: Connect power module.");
		feedback_provided = true;
		valid_transition = false;
		}
----------------------------------------------------------------------------------------------------------

tubeme, 2017-03-13:
@bssung You get this message because you did not use the Power Module connected to Pixhawk which looks like this:

http://www.ebay.com/itm/New-Crius-Power-Module-28V-90A-for-Pixhawk-APM2-8-APM2-6-APM-Flight-Controller-/281220147710

There are some other power module options as well.

This module has 2 functions: 1. to give the right 5v to the Pixhawk board. 2. It is a sensor for the voltage of the battery and it senses the AMP consumption of the system. 

Without this information you are flying blind in terms of consumption of electricity in the system so the system needs the power module in order to fly consistently. If you have the power module connected and you still get this message, then it means there is some wrong wiring that you did.

Here is a connection diagram:

https://pixhawk.org/users/tutorials/pixhawk_6s_mod

bssung, 2017-03-14:
Thank you very much!

closed 2017-04-03 by mhkabir

krunal2103, 2017-04-15:
@tubeme , I had the same problem of the pixhawk not getting armed. I had also used the power module to power the pixhawk. @bssung , can you just help me with what exactly did you do to solve this problem?

bssung, 2017-04-17:
@krunal2103 , you need to refer to the log file in sd card, get the detail message.

niranjanstudy06, 2021-09-26:
**Pixhawk only arms when connected to QGC via wired/wireless telemetry.** 

Hello, I have installed the Pixhawk firmware using QGC. Calibrated all sensors and everything. It arms perfectly when connected to the QGC via laptop MicroUSB cable. But when I disconnect the cable and try to arm it. It doesn't arm the drone. Why is this and how to fix it.

niranjanstudy06, 2021-09-27:
@tubeme @bssung  Kindly Help me out in solving this problem. I have checked many forums but am not able to solve the problem.

JMoore5353, 2022-09-18:
@niranjanakella  I know this is almost a year later, but I had the same problem as you and figured out the problem. Hope this helps if anyone is looking for an answer.

For reference, my problem was as follows: 
I set up my Pixhawk 2.4.8 with QGC, calibrated all the sensors and then tried to arm. The motors would arm when I was connected to the pixhawk via USB (microUSB) to my computer. But when I tried to arm without the USB connection, I would get an error (the pixhawk would flash red 3 times and wouldn't arm).

I tried disabling the safety switch and all sorts of things, but I finally checked the logs and discovered the COM_POWER_COUNT param. This checks for a specified number of redundant power sources. Mine was set to 2 as default, so with the USB and battery, the px had 2 power supplies, so it passed that parameter, but without the USB, it would fail to arm.

I ended up having to disable CBRK_SUPPLY_CHK, so it wouldn't even check for that when arming. I will go back and see if I can get it to work when CBRK_SUPPLY_CHK is enabled.

Hope that helps!

hagen6835, 2022-09-21:
> m

This helped me a ton!! I was having this exact issue and the fact that you posted the solution 3 days earlier is incredible timing haha. Thanks!

Ranamajdi99, 2023-02-01:
I HAVE BEEN STRUGGLING WITH THAT FOR 2 DAYS, UNTIL I FOUND YOUR SOLUTION!! THANKS A LOT!!!