PX4/PX4-Autopilot #6773: PWM signals not coherent in offboard mode

marcelinomalmeidan, 2017-03-10:
Hi everyone,

We have a Snapdragon Flight here with Qualcomm's ESC board. We calibrated the motors and we've seen them spin after the calibration procedure (so we think that the ESC board was calibrated correctly).
After that, we tried to run MAVROS with offboard mode as [shown here](https://dev.px4.io/ros-mavros-offboard.html) . We are sending thrust commands to the vehicle (from 0 to 1) and reference quaternions.
When running in SITL simulation (gazebo), it works fine. When we monitor /mavros/rc/out (still in simulation) we can see the PWM values that are being sent to each motor.
However, in the snapdragon flight, the topic /mavros/rc/out prints only 0 and 1 (binary), which I don't understand. I would expect values from 1000 to 2000, which are the calibrated values for the PWM signals. ([see image](https://www.dropbox.com/s/tdh4ktmoh3kwk05/screenshot_from_2017-03-10_12-52-52.png?dl=0)).

LorenzMeier, 2017-03-13:
@jywilson Are you able to help?

In general we're not supporting the Qualcomm ESC board officially but PWM motor controllers. If Qualcomm wants to change that we would need a couple of test units, which we have not received to date.

jywilson, 2017-03-24:
Communication with the ESC board occurs over one UART link to all four ESC's.  You need to use the ESC driver that is included in the FCAddon which can be downloaded from the Intrinsyc website.  The ESC drivers available in PX4 will not work with this ESC.  

Look at the px4.config file in the 210qc directory for an example of how to load to load this driver.

Jim W.

dagar, 2017-04-11:
@marcelinomalmeidan is this resolved?

marcelinomalmeidan, 2017-04-12:
@dagar
I guess my question didn't make much sense. Mavros doesn't publish anything to rc/out because the outputs of snapdragon flight are not directly PWM, they are UART.

closed 2017-04-12 by marcelinomalmeidan