PX4/PX4-Autopilot #6778: Snapdragon Flight parameter synchronization is broken

jywilson, 2017-03-11:
The qurt and posix builds generate separate parameter.xml files.  Since the current parameter synchronization mechanism uses an index to update the parameter representation in memory on the aDSP and AppsProc, attempting to update a parameter in the AppsProc will cause the wrong parameter to be updated in the aDSP.

Stifael, 2017-03-11:
https://github.com/PX4/Firmware/issues/6261

LorenzMeier, 2017-03-11:
@jywilson @mcharleb We need the params on both sides to be synced. That's super important because anything else is very non-intuitive.

bkueng, 2017-03-13:
@jywilson the different parameters.xml can come from the [cmake scoping](https://github.com/PX4/Firmware/blob/master/cmake/common/px4_base.cmake#L1081) if the two builds use different modules.

jywilson, 2017-03-15:
@LorenzMeier I agree.  This is probably responsible for the QGC crash reported here: https://github.com/mavlink/qgroundcontrol/issues/4738

More comments on this problem here: https://github.com/PX4/Firmware/issues/6765

jywilson, 2017-03-21:
I am working top resolve this issue now.  

@LorenzMeier Can you assign it to me?  I can't seem to find a way to assign this to myself on my own.

jywilson, 2017-03-24:
This issue is fixed in this PR:
https://github.com/PX4/Firmware/pull/6883

jywilson, 2017-03-28:
A fix for this issue is now on master.

closed 2017-03-28 by jywilson