PX4/PX4-Autopilot #6788: Some questions after a crash..

BossKwei, 2017-03-12:
http://logs.uaventure.com/view/gTQjyURYX6LvADB7KcYYYV

Firmware: 1.5.5 Stable

# Crash review
1. Arming the quadrotor. (in POSCTL mode)
2. Push the throttle forward for a little, then four rotors began to spin, everything is fine.
3. Disarming the quadrotor.
4. Next arming again without a power off. (also in POSCTL mode)
5. Then when I push the throttle, **the 1st rotor did not spin**, my quadrotor crashed and then rollover.
6. I tried to disarm it but nothing happened.

# Questions
## 1.  PX4 does not respond to my disarming operation, why? (Is there a bug in px4 firmware?)
In **Pitch Angle** diagram, my quadrotor crashed and then rollover.
In **Raw Radio Control Inputs** diagram, I tried to disarm it but nothing happened.. Then I had to cut off the power.

## 2. Is px4flow in use with ekf2 estimator? 
I installed a px4flow to my quadrotor, but there's nothing in **Distance Sensor** diagram, does this sonar work in ekf2 estimator or is there any param like SONAR_EN? 

## 3. Why could not Land in POSCTL Mode?
It is read that "Do not Land in POSCTL Mode, use Manual Mode to Land." in User Guide, I am really puzzled about this, why?

## 4. Is there a param used for disarming quickly?

LorenzMeier, 2017-03-12:
Here is what happened: Your ESCs are incorrectly calibrated, which means that the first motor (http://logs.uaventure.com/view/gTQjyURYX6LvADB7KcYYYV#Actuators_PLOT) was not actually spinning when it should have.

That led to the system tipping over. You are not allowed to disarm in position control mode because lowering the altitude and yawing is a common user behaviour - however, if you do that, you would disarm the system mid-flight. You can configure a kill switch, which will kill the outputs instantly.

To recap: To not have this happen again, calibrate your ESCs or set the PWM_MIN parameter for the main outputs to a higher value.

closed 2017-03-12 by LorenzMeier

BossKwei, 2017-03-12:
@LorenzMeier Thank you for your reply, and I'm really sorry about this issue, there isn't the appropriate place for configuration fault.

BossKwei, 2017-03-12:
@LorenzMeier And, how could you find my fault? Does any diagram show this?