PX4/PX4-Autopilot #6790: Logging rates and topic instances in new logger

AndreasAntener, 2017-03-12:
@bkueng in reference to https://github.com/PX4/Firmware/pull/6770/commits/135f64aa7d54cf93e2417c0e85e618e9fe25c44f

I'd like to make a PR with some improvements, could you help me with the following:

- Is it correct that logging of difference instances is not supported yet? We need `actuator_outputs` 1 and 2
- The rates for attitude and setpoints, as well as position and setpoints should be the same
- Why is the `position_setpoint_triplet` logged at such a high rate?

LorenzMeier, 2017-03-12:
* Different instances are supported, selecting the topic will log all
  * Fixed in e4c012289f
  * Fixed in f7e4d8453d

AndreasAntener, 2017-03-12:
Thanks!

Regarding topic instances: I only see `actuator_outputs_0` in our VTOL logs.

LorenzMeier, 2017-03-12:
This is not a VTOL but shows things are working:

[image]

Actuator controls are a slightly different beast - they are not multi-topics.

LorenzMeier, 2017-03-12:
I'm closing this - I'm sure we need to further fine-tune .ulog before it works for all use cases, but I think its best to open new issues for specific things we find, rather than keeping a monster issue around.

closed 2017-03-12 by LorenzMeier