PX4/PX4-Autopilot #6791: a error when make px4esc-v1_default

zouboan, 2017-03-12:
when i build the firmware of px4esc-v1 using:
make px4esc-v1_default
there's a error:

make[3]: *** No rule to make target `src/firmware/nuttx/firmware_nuttx.bin', needed by `org.pixhawk.px4esc-v1-1.6-0.1.8c9f27bb.uavcan.bin'.  Stop.
make[2]: *** [CMakeFiles/build_uavcan_bl_px4esc-v1.dir/all] Error 2
make[2]: *** Waiting for unfinished jobs....
[  1%] Built target git_nuttx
[  1%] Built target git_genmsg
[  2%] Built target git_gencpp
make[1]: *** [all] Error 2

I am puzzled,and  looking forward to be disabused,thanks!

zouboan, 2017-03-12:
When i remove following code：

px4_nuttx_make_uavcan_bootloadable(BOARD ${BOARD}
 BIN ${CMAKE_CURRENT_BINARY_DIR}/src/firmware/nuttx/firmware_nuttx.bin
 HWNAME ${uavcanblid_name}
 HW_MAJOR ${uavcanblid_hw_version_major}
 HW_MINOR ${uavcanblid_hw_version_minor}
 SW_MAJOR ${uavcanblid_sw_version_major}
 SW_MINOR ${uavcanblid_sw_version_minor})

the make can be passed,why?
and what does it affect after I delete it?

LorenzMeier, 2017-03-12:
Run `make distclean` (and ensure to back up your work before).

closed 2017-03-12 by LorenzMeier

zouboan, 2017-03-12:
Run make distclean and then make px4esc-v1_default
the error still appears after

leomatrix, 2017-05-26:
I have the same problem....