PX4/PX4-Autopilot #6792: PX4 SITL running busy

LorenzMeier, 2017-03-12:
@bkueng SITL runs busy on master. I don't have named threads yet, but one of them seems to just run in a tight loop.

[image]

LorenzMeier, 2017-03-12:
Commit 01541bb10d98f17c44cb11a118bba3df49d5ddd6 is affected (March 3). Commit 51c8e905085cab9fe73f204b0fcdd0c43130c7a2 is affected as well.

LorenzMeier, 2017-03-12:
Similar results in Gazebo (above is jMAVSim):

```
Threads: 23 total
thread 0		 0
thread 1		 441
thread 2		 2
thread 3		 0
thread 4		 14
thread 5		 0
thread 6		 30
thread 7		 0
thread 8		 0
thread 9		 4
thread 10		 22
thread 11		 10
thread 12		 11
thread 13		 21
thread 14		 12
thread 15		 24
thread 16		 58
thread 17		 0
thread 18		 51
thread 19		 0
thread 20		 27
thread 21		 3
thread 22		 0
pxh> 
```

LorenzMeier, 2017-03-12:
Getting the thread name is quite convoluted. Making the assumption that these are in-order (which is most likely true:

```
pxh> list_tasks
INFO  [platforms__posix__px4_layer] Active Tasks:
INFO  [platforms__posix__px4_layer]    hpwork     123145381175296
INFO  [platforms__posix__px4_layer]    lpwork     123145381711872
INFO  [platforms__posix__px4_layer]    wkr_hrt    123145382248448
INFO  [platforms__posix__px4_layer]    dataman    123145382785024
INFO  [platforms__posix__px4_layer]    simulator  123145383321600
INFO  [platforms__posix__px4_layer]    gps        123145384394752
INFO  [platforms__posix__px4_layer]    pwm_out_sim 123145384931328
INFO  [platforms__posix__px4_layer]    sensors    123145385467904
INFO  [platforms__posix__px4_layer]    commander  123145386004480
INFO  [platforms__posix__px4_layer]    navigator  123145386541056
INFO  [platforms__posix__px4_layer]    ekf2       123145387077632
INFO  [platforms__posix__px4_layer]    mc_pos_control 123145387614208
INFO  [platforms__posix__px4_layer]    mc_att_control 123145388150784
INFO  [platforms__posix__px4_layer]    mavlink_if0 123145388687360
INFO  [platforms__posix__px4_layer]    mavlink_if1 123145389760512
INFO  [platforms__posix__px4_layer]    logger     123145390833664
```
and

```
Threads: 23 total
 0 (775) 		 0
 1 (2819) 		 725
 2 (3331) 		 3
 3 (3587) 		 0
 4 (3843) 		 13
 5 (4099) 		 0
 6 (4355) 		 35
 7 (4611) 		 0
 8 (4867) 		 0
 9 (5123) 		 4
 10 (5379) 		 20
 11 (5635) 		 7
 12 (5891) 		 0
 13 (6147) 		 18
 14 (6403) 		 6
 15 (6659) 		 21
 16 (6915) 		 35
 17 (7171) 		 0
 18 (7427) 		 36
 19 (7683) 		 0
 20 (7939) 		 21
 21 (8195) 		 1
 22 (9475) 		 0

total: 20 sec, 4805028 usec, 945 cpu
```

I'm not sure what to conclude yet.

bkueng, 2017-03-13:
I cannot reproduce. My SITL uses ~11% of one core when flying with jMAVSim. What's the meaning of the numbers?

dennisss, 2017-03-26:
Also had this issue. On macOS specifically. The Instruments: Time Profiler is pretty useful was pretty handy for finding this issue:
[image]

See PR above for fix.

closed 2017-03-26 by LorenzMeier

LorenzMeier, 2017-03-26:
@dennisss Thanks! Fix is on master.