PX4/PX4-Autopilot #6794: Error when calling px4_nuttx_make_uavcan_bootloadable error

zouboan, 2017-03-12:
If run：
make esc35-v1_default
or
make px4esc-v1_default
or
make s2740vc-v1_default
or
make px4cannode-v1_default

there will coming a same error:
make[3]: *** [  1%] No rule to make target `src/firmware/nuttx/firmware_nuttx.bin', needed by...

and if remove the calling of px4_nuttx_make_uavcan_bootloadable in the ./cmake/configs/*.cmake,the bilud will be passed!

davids5, 2017-04-03:
I think this may be related to your build setup. 

This builds fine https://semaphoreci.com/px4/firmware/branches/master/builds/1221 see ./Tools/docker_run.sh 'make alt_firmware'

Please run ```make clean``` and ensure you have all the submodules updated (make distclean) also check the version of the compiler it should be 5.4.1

dagar, 2017-04-08:
@zouboan is this resolved?

davids5, 2017-04-22:
@zouboan I am closing this. If it still an issue please feel free to reopen

closed 2017-04-22 by davids5