PX4/PX4-Autopilot #6796: UAVCAN ESC don't seem to respond the way they should

AndreasAntener, 2017-03-12:
http://logs.px4.io/plot_app?log=155832f2-b37e-44ae-a81e-fb9c8dd5de02

[image]

Using: https://github.com/PX4/Firmware/pull/6795

This log shows a VTOL flight where the vehicle lost attitude and height control after the back transition. Observations:
- power consumption of the whole system is very low after the back transition even though the thrust setpoint goes to full
- the feedback from the ESCs suggest they are doing something but also that for example one motor is still turned off
- attitude tracking and performance in general is fine at takeoff and before the front transition
- the UAVCAN ESC status is shifted in time and logged ~2 seconds too early (or the rest too late)

LorenzMeier, 2017-03-12:
Were you flying with Sapog or PX4esc? Those are two different codebases. It seems like Sapog might have had spinup issues:

https://github.com/PX4/sapog/pull/22

LorenzMeier, 2017-03-12:
@pavel-kirienko Can you chime in?

AndreasAntener, 2017-03-12:
Hm, not sure, the quad setup from Michael/Mathieu?
This happened now twice in flight, but never on the bench (did 10 bench tests at least today)

LorenzMeier, 2017-03-12:
The quad setup is Sapog. Can you upgrade them and re-test? I would not be surprised at all if this is related.

LorenzMeier, 2017-03-12:
And do you have bandwidth to look into the logging issue?

AndreasAntener, 2017-03-12:
> - can pick up synchronization if the rotor is already rotating.
>
> Significantly more reliable operation during rapid acceleration and deceleration, especially at high advance angles.

Well that sounds very suspicious..

pavel-kirienko, 2017-03-12:
@AndreasAntener Is it true that you were restarting the motors in flight (this is what I'm reading from the plots)? If so, then indeed, it could be caused by the spinup issues I've been fixing in the last few days. You might find the new beta to perform noticeably better.

AndreasAntener, 2017-03-12:
Michael and I figured out the logging issue today, fix incoming.

AndreasAntener, 2017-03-12:
Sorry my bad, I guess you mean the timing issue, no not that (but the multi instance problem which is not logging related)

AndreasAntener, 2017-03-12:
@pavel-kirienko yes that is true. Which would explain why we cannot reproduce it on the bench. What's the easiest to upgrade (sorry I'm a UAVCAN newbie here)

pavel-kirienko, 2017-03-12:
@AndreasAntener this should help: https://dev.px4.io/uavcan-node-firmware.html#firmware-installation-on-the-autopilot

AndreasAntener, 2017-03-12:
Thanks. Stupid question: what hardware version do I have? :)

pavel-kirienko, 2017-03-12:
You have v1.0 I guess.

pavel-kirienko, 2017-04-24:
@AndreasAntener fixed in https://github.com/PX4/sapog/pull/22. Version 2.0 should be officially released before the end of the week.

pavel-kirienko, 2017-04-26:
Should we close it, or is anybody willing to volunteer and test it? My own tests indicate that the issue is fixed. @mhkabir @AndreasAntener please advice.

mhkabir, 2017-04-26:
@eyeam3 Will be testing the latest Sapog release within the next few days.

bresch, 2017-06-07:
@pavel-kirienko @AndreasAntener Tested today with Orel20 and Sapog v2.2. I first tried if my drone was able to recover from a free-fall and then I tried to flip the kill switch during ~0.5s. The motors were able to restart successfully.

[Video](https://drive.google.com/open?id=0B0uX_2YN9jiNeUVNbU1OVENfVm8) of the kill switch flip sequence and complete [log](http://logs.px4.io/plot_app?log=6c80a1d7-6127-4948-a0f3-c65e92b86bcb).

Below, the graph of the current consumption of each ESC:
[image]

closed 2017-06-07 by bresch

pavel-kirienko, 2017-06-07:
Thanks, Mathieu.