PX4/PX4-Autopilot #6798: Yaw estimate jumps

AndreasAntener, 2017-03-13:
http://logs.uaventure.com/view/UKyU98TtEzD9iM7aS6JErg
[image]

At ~23.5 the estimated heading jumps by ~20 degrees. This happened multiple times over different hover flights, but only once per flight as far as I can see. Any thoughts?

Setup: Pixhawk Pro, Zubax GNSS 2.0

AndreasAntener, 2017-03-13:
Post flight: https://gist.github.com/AndreasAntener/f45bb496c9e6d41a197360ed16d2f9f0
I see a lot of buffer overruns for the internal sensors, but I don't see any info about the external mag. Also it seems messages were not written to file.

LorenzMeier, 2017-03-13:
The buffer overruns are irrelevant. I probably should remove those because we're not using that interface.

LorenzMeier, 2017-03-13:
Closing this in favour of https://github.com/PX4/Firmware/issues/6803, which is the root cause.

closed 2017-03-13 by LorenzMeier

gijimbo56, 2017-05-09:
hhhmmm, well I cant speak to the firmware for the GNSS in the post you mention #6803 . I can say that I have seen this elsewhere with yaw jumps related to magnetometers. 

Here is the link to the other forum that has been helpful. Given their comments wouldn't call this issue closed. If this is the case. 

http://discuss.px4.io/t/huge-yaw-movement-everytime-i-take-off/2982/13

We just had a big issue with this on one of our quads in stabilize mode. It appears there is change in heading estimation (how the mags are used) after a certain height to account for mag interference. There doesnt seem to be check in the PX4 code to see if the yaw jump is reasonable after a heading estimation change, after a certain altitude is met. This could then cause a jump in the command to the motors which we saw in our experiments and thus a yaw jump in the vehicle. Maybe I am wrong...but it seems to be the issue. 

Thanks for any feedback that can be provided.