PX4/PX4-Autopilot #6799: Pixhawk Pro safety switch pressed state changes randomly

AndreasAntener, 2017-03-13:
When no safety switch is connected the pressed state changes randomly. As a workaround I've disabled the GPIO handling and forced the state to false.

klopezal, 2017-03-13:
@AndreasAntener 

`GPIO_BTN_SAFETY` is defined twice, once in board_config.h (Pixracer style, because it has no IO processor) and once in IO driver. I would try commenting [this line](https://github.com/PX4/Firmware/blob/06c5cb506a6bcb623fde88f9b15df646f1dfd9ec/src/drivers/boards/px4fmu-v4pro/board_config.h#L301) and [this line](https://github.com/PX4/Firmware/blob/06c5cb506a6bcb623fde88f9b15df646f1dfd9ec/src/drivers/boards/px4fmu-v4pro/px4fmu_init.c#L197).

klopezal, 2017-03-13:
@AndreasAntener @davids5 

Changing [this line](https://github.com/PX4/Firmware/blob/master/src/drivers/boards/px4io-v2/board_config.h#L82) by 
`#define GPIO_BTN_SAFETY (GPIO_INPUT|GPIO_CNF_INPULLDWN|GPIO_MODE_INPUT|GPIO_PORTB|GPIO_PIN5)` 
solves the issue. David, is that a good way of dealing with it or should we implement a hardware pull-down resistor?

davids5, 2017-03-13:
@klopezal - I have a PR inbound. 
We need to look for this same class of mistake in the board_config that are a byproduct of the copy from fmuV4.

I found one more it is in the incoming PR

davids5, 2017-03-13:
@AndreasAntener @davids5
See https://github.com/PX4/Firmware/pull/6808

AndreasAntener, 2017-03-14:
@davids5 Thanks! I don't fully understand the fix though.

For my understanding: `GPIO_BTN_SAFETY` is still referenced in `px4io.h`. Wouldn't that also require changing? Also I'm a bit confused since `GPIO_BTN_SAFETY_FMU` now only exists for v4pro.

davids5, 2017-03-14:
There are two issues.
The first is that an unconnected signal had the same name as the button input. By changing its name that disables the fmu from treating that input as the safety switch.

The second issue is that the actual safety switch connection to the PX4 IO is missing a pull down. The proper fix is to add the pull-down on the Hardware. An alternate workaround is to use a modified bootloader and firmware build that both enables  pull down  on that gpio. 

In this case because the signal is at the edge of the board and the pins are large I would recommend soldering through hole component on it and connecting it to ground. It needs to be 1.5 k

davids5, 2017-05-17:
>The first is that an unconnected signal had the same name as the button input. By changing its name that disables the fmu from treating that input as the safety switch.

Fixed In FW

closed 2017-05-17 by davids5