PX4/PX4-Autopilot #6819: LPOS.Z didn't change along with LPOS.VZ

liang-tang, 2017-03-15:
something wrong with kf2 estimator, LPOS.Z didn't change along with LPOS.VZ?
can anybody analyze this log
[image]
[image]

http://logs.uaventure.com/view/En3ZH5CkCMLTyuMByzw6Eb

liang-tang, 2017-03-15:
@julianoes

priseborough, 2017-03-15:
The accel bias estimation is not converging. Vertical vibration levels are high. Please provide details of your test setup and a log containing EKF replay data.

You can turn off accel bias learning by setting EKF2_AID_MASK = 5 but will need to make sure the accelerometers have a recent calibration if you do and may have to increase the value of EKF2_ACC_NOISE to 0.5 m/s/s to allow for lack of bias compensation.

liang-tang, 2017-03-15:
thank you very much, i'll do more test tomorrow.

mhkabir, 2017-04-03:
@liang-tang Please provide logs, or we will close this soon.