PX4/PX4-Autopilot #6823: SITL local position not getting reset when I set home position

lovettchris, 2017-03-15:
If you change the LatLonAlt default in jMavSim Simulator.java to Seattle instead of Zurich, then run jMavSim, you will find that MAVLINK_MSG_ID_LOCAL_POSITION_NED reports x=7170790.000000, y=-4435566.500000, z=0.000490.  it is 8442 kilometers to Zurich, so this indicates that local position is still trying to compute distance back to Zurich, because that is the "global origin init" 

even after I arm the drone or send CMD_DO_SET_HOME 

I would have expected we would want local coordinates to always be relative to the drone launch point, especially given the inaccuracyof the conversion from lat/lon to linear x,y,z.  

I don't see local_position_estimator or ekf2 listening to set_home?  I do see the Commander calling commander_set_home_position when the drone is armed, but the local_position is not reset to 0,0,0.

Oddly enough ekf_att_pos_estimator_main does subscribe to ORB_ID(home_position) but it doesn't actually use that subscription.  mc_pos_control uses it, but only to help with landing, and the navigator uses it, but only for geofence.  

I added an additional print statement in the Commander to make this more visible.  When SITL starts up you see this:
INFO  [commander] home gps: 47.3977432, 8.5455942, 103.72
INFO  [commander] home local: -0.0000122, 0.0000065, -0.00

but when jMavSim sets the new home position you see this:
INFO  [commander] home gps: 47.5921230, -122.3160651, 103.48
INFO  [commander] home local: 7170789.0000000, -4435566.0000000, 0.01

So am I missing something here? If not, what is the recommended fix that I can try out?

lovettchris, 2017-03-15:
Interesting, I found the code I was expecting, in BlockLocalPositionEstimator, but all commented out:

	// reset pos, vel, and terrain on arming

	// XXX this will be re-enabled for indoor use cases using a
	// selection param, but is really not helping outdoors
	// right now.

	// if (!_lastArmedState && armedState) {

	// 	// we just armed, we are at origin on the ground
	// 	_x(X_x) = 0;
	// 	_x(X_y) = 0;
	// 	// reset Z or not? _x(X_z) = 0;

	// 	// we aren't moving, all velocities are zero
	// 	_x(X_vx) = 0;
	// 	_x(X_vy) = 0;
	// 	_x(X_vz) = 0;

	// 	// assume we are on the ground, so terrain alt is local alt
	// 	_x(X_tz) = _x(X_z);

	// 	// reset lowpass filter as well
	// 	_xLowPass.setState(_x);
	// 	_aglLowPass.setState(0);
	// }

lovettchris, 2017-03-15:
but that doesn't work either.  Honestly I was expecting something more like this:

	if (_sub_home.updated()) {
	 	// home was just changed, so reset local position to 0,0,0
	 	_x(X_x) = 0;
	 	_x(X_y) = 0;
		_x(X_z) = 0;

		mavlink_and_console_log_info(&mavlink_log_pub, "[lpe] home position changed, resetting local position.");

		home_position_s home = _sub_home.get();

		map_projection_init(&_map_ref,
			home.lat,
			home.lon);

		_altOrigin = home.alt;

		mavlink_and_console_log_info(&mavlink_log_pub, "[lpe] reset global origin init (parameter) : lat %6.2f lon %6.2f alt %5.1f m",
					     double(home.lat), double(home.lon), double(_altOrigin));
	}

mhkabir, 2017-03-15:
The origin isn't supposed to be reset when setting home. The origin should always be fixed.

lovettchris, 2017-03-15:
But it doesn't work, if jMavSim sets initial position to Seattle then QGroundControl can't take off.  First QGroundControl  is all jittery (I think it is jumping quickly between two very different locations), then when you tell it to take off the drone flies sideways into the ground.

Besides it is really inconvenient to have to program all local positions with really big numbers.  I want to have a portable flight script that I can run anywhere using local coordinates starting at 0,0,0 no matter where I put the drone and take off.  This seems to be the case with my real Pixhawk...  Just seems SITL works differently for some reason...

lovettchris, 2017-03-15:
Doing more debugging I found the wrong GPS location is coming from a parameter named LPE_FAKE_ORIGIN.  When I clear that, then SITL never sets any home location at all, even when jMavSim is sending lots of HIL_GPS messages, which seems odd.  So looks like SITL mode requires the use of LPE_FAKE_ORIGIN...  When I set this to Seattle location, jMavSim behaves better...  and QGroundControl is happy to fly around Seattle.  So I guess it is just one more step in getting SITL mode setup correctly, which makes it more of a hassle.  Would be nice if it determined the origin the same way HITL mode does (namely, from HIL_GPS).

mhkabir, 2017-03-16:
You shouldn't need to have fake origin set at all, because it's used for optical-flow only navigation. 

The reason it doesn't init after disabling FAKE_ORIGIN is because of vision being enabled.  Make sure vision position fusion flags are turned off (LPE_FUSION bitmask in QGC). It won't init home with GPS if vision is enabled. 

@jgoppert I think we need better defaults for master, in terms of FAKE_ORIGIN and LPE_FUSION to avoid user issues like these.

jgoppert, 2017-03-16:
Yeah, the fake origin change should have gone through more rigorous testing.  I think the default should be for outdoor with gps and no vision or flow. Can we confirm that works and then submit a PR?

mhkabir, 2017-03-16:
I tested it in all configs (GPS, VIO). It works fine, however, we just need better defaults. I didn't change the behaviour in my PR - so this "issue" was present even before my FAKE_ORIGIN addition.

I'll send in a PR with the GPS-default parameter set.

jgoppert, 2017-03-16:
Awesome, thanks.

lovettchris, 2017-03-21:
paste a link to your PR so I can test it out?  can you also tell me what the "design" is for setting local position origin?  I thought origin was the point you armed the drone so it can always return to that launch poiunt (RTL and all that).  Am I wrong?

mhkabir, 2017-03-21:
@lovettchris Sorry, haven't got around to doing it yet. Will have it by the end of this week. 

The _origin_ is the point where the estimator initialises (0,0,0). What you're talking about is the _home position_, where it RTLs to, and is reset on arming.

mhkabir, 2017-03-25:
Fixed here : #6897.
@lovettchris Can you please cross-test.

PR #6897 (2017-03-26): LPE : default fusion flags for GPS-only use
This changes the LPE fusion flags to default to GPS-only mode. For optical flow based nav, I will add a Devguide entry on how to set it up properly : https://github.com/PX4/Devguide/pull/98

Fixes #6823. @lovettchris You will want this :)
closed 2017-03-26 by LorenzMeier