PX4/PX4-Autopilot #6824: cant enter the nsh mode

potato77, 2017-03-15:
i follow the devguide and enter "screen /dev/ttyACM0 BAUDRATE 8N1"
and it shows a new console. but i cant enter any words.

[image]

How to fix it? Thank you in advance!

any help would be appreciated!

d3z0rian, 2017-03-15:
Did you replaced the "BAUDRATE" word for the board baudrate in number?
Serial: Pixhawk v1 / Pixracer use 57600 baud
Serial: Snapdragon Flight uses 115200 baud

potato77, 2017-03-16:
@d3z0rian

yes

mhkabir, 2017-03-16:
Since this is a VM, there may be issues with accessing the port on the host machine. Not sure how we can help with this directly.

mhkabir, 2017-04-03:
If this still persists, try removing modemmanager, and making sure your user is in the dialout group.

closed 2017-04-03 by mhkabir