PX4/PX4-Autopilot #6826: SITL + gazebo

AlexMalo, 2017-03-15:
I am trying to make a sitl simulation of a plane and I am using gazebo (since X-plane is huge), but it seems that the computer/simulation is too slow, I wonder if the gazebo simulation be done only with gzserver (withoug the graphics)

L4ncelot, 2017-03-15:
Hi,

I'm assuming you're using ROS in this setup. Therefore in the launch file where your Gazebo simulator starts, you can set argument named **gui** to **false**.

Example of such a file would be something like this:
```
<launch>
<arg name="gui" default="true"/>
    <include file="$(find gazebo_ros)/launch/empty_world.launch">
        <arg name="headless" value="$(arg headless)"/>
        <arg name="gui" value="$(arg gui)"/>
        <arg name="world_name" value="$(arg world)"/>
        <arg name="debug" value="$(arg debug)"/>
        <arg name="verbose" value="$(arg verbose)"/>
        <arg name="paused" value="$(arg paused)"/>
    </include>
</launch>
```

dagar, 2017-04-11:
@AlexMalo if launching from the PX4 build system you can also set the environment variable HEADLESS.

    HEADLESS=1 make posix_sitl_default gazebo_plane

closed 2017-04-11 by dagar

AlexMalo, 2017-04-27:
The ROS solution is cumbersome, maybe because I am not familiar with ROS; I tried it but hit some compatibility issues. It added another layer. The headless parameter works for me. Thank you.