PX4/PX4-Autopilot #6827: FLAPS: Does not work in 1.6.0 RC1

tubeme, 2017-03-15:
Hi guys,

I'm just setting up a v-tail airplane with Albatros mixer and the Flaps does not work. I preset RC channel for flaps and nothing happens when switched.

The flaps are working because I test it with servo tester but no functionality from Main Out 7,8.

tubeme, 2017-03-16:
I tried in Manual and Stabilized on the bench... Nothing happens. Before there was a problem with flaps not fully extending https://github.com/PX4/Firmware/issues/4438. But now they just don't move at all.

tubeme, 2017-03-22:
To add to the problem I tried to make a workaround for the Flaps and use Aux1 and Aux2 with RC pass-through to set it up with no luck.

There is no RC pass-through with this mixer lines:

Flaps mixer
------------
Flap servos are physically reversed.

M: 1
O:      10000  10000      0 -10000  10000
S: 3 5      0   5000 -10000 -10000  10000

M: 1
O:      10000  10000      0 -10000  10000
S: 3 6      0  -5000  10000 -10000  10000

Nothing moves! I see there are Aux1 and Aux2 input to PX4 because in QGC the sliders move. But nothing is passed to the servos.

Probably there is something wrong with the mixer in general.

see3peoh, 2017-05-10:
same issue here - also not getting my nose wheel moving either - going back to stable and all works fine. probably something to do with mixer size which was an issue a long time back. were you able to make a work around?

tubeme, 2017-05-10:
@see3peoh I was able to control the nose wheel, this might be something more recent. But the flaps still does not work neither the rc forwarding. The workaround is you use a spare channel on the RC receiver... but you cannot make automation this way except you make the automation in the transmitter mixer, in my case Taranis.

LorenzMeier, 2017-05-10:
Did you test with current master?

see3peoh, 2017-05-10:
@tubeme  Ok - thanks. Too bad this still isn't working. Thanks for the workaround but chaos for mission auto. Likely something with current master re the nose gear. Loaded up stable and everything worked again just to check. Also on a Taranis.

@LorenzMeier Yes, no go.

see3peoh, 2017-05-10:
How does one systematically look through commits to ascertain which/what broke this? If there isn't a methodical approach, I'm happy to just jump large amounts backwards (of commits to RC1), build and test but something more targeted would be ideal. This is actually more of an issue than I thought as without flaps it causes some takeoff/landing concerns but without the nose wheel, there is no real active auto mission that is working.

LorenzMeier, 2017-05-10:
There is this one: https://git-scm.com/docs/git-bisect but I suspect that its rather the result of an architectural cleanup on the PWM outputs. Can you provide this output:

```
pwm info -d /dev/pwm_output0
```
and
```
pwm info -d /dev/pwm_output1
```
and
```
px4io status
```

You can go in QGC to the log download tab on the far right and start the MAVLink shell there to launch these commands via USB.

LorenzMeier, 2017-06-04:
Ping - can anyone provide these outputs with an armed system? I cannot debug with no data.

tubeme, 2017-07-07:
In PX4 1.6.3 flaps are working perfectly well. But I had to make my own mixer rot the AETR airplane. The same holds for the AERT mixer. Both mixer files lack the line:

Flaps  mixer for CH5,
using the payload control group
-----------------------------------------------------

M: 1
O:      10000  10000      0 -10000  10000
S: 0 4  10000  10000      0 -10000  10000

Once I added this line to the mixer the flaps were alive.

LorenzMeier, 2017-10-30:
Should be fine as we merged the definitive fix recently.

closed 2017-10-30 by LorenzMeier

koenlucassen, 2020-01-28:
I am actually facing a similar issue with the current Px4 version (1.10.1). I want to control the flaps via control group 0 channel 4 with the Bormatec Maja airframe (thus main 7 channel).