PX4/PX4-Autopilot #6834: RC controller in HIL configuration

L4ncelot, 2017-03-16:
Hi,

I'm working with px4 firmware on **navio2**. Right now I'm testing **HIL** configuration where **navio2** board is connected through ethernet to my PC with **Gazebo** simulator. The configuration works fine and I'm able to control quadcopter sending **MAVLINK** messages through **MAVROS** (yes, I'm also using **ROS**).

What I would like to do is connect my RC controller to navio2 board and control quadcopter with it. I had and idea where I would create simple program to read from RC channels located in `/sys/kernel/rcio/rcin` and transfer them to **MAVLINK** messages which would be send to px4. But I don't wanna write something on my own if there's already support for it.

From [this](https://github.com/PX4/Firmware/blob/master/Documentation/px4_hil/SITL_Diagram.png) diagram found in documentation it seems like there should be some kind of support but I'm kind of lost in px4 code.

So I would appreciate some kind of hint on how to work with RC controller.

Thank you.

bkueng, 2017-03-16:
There's already a driver for the navio2 RC: https://github.com/PX4/Firmware/blob/master/src/drivers/navio_sysfs_rc_in/navio_sysfs_rc_in.cpp
If you start that px4 should automatically handle RC.
I recently added a HIL config for the Pi: https://github.com/PX4/Firmware/blob/master/posix-configs/rpi/px4_hil.config. It works with jMAVSim (I haven't tried RC though), but it should help you with gazebo as well.

L4ncelot, 2017-03-16:
Thank you very much, I will have a look at it.

L4ncelot, 2017-03-16:
Oke so after a while I figured it out. Adding drivers/`navio_sysfs_rc_in` target to **posix_sitl_default.cmake** file.

So I managed to run the firmware with `navio_sysfs_rc_in`. Now what should I do next to be able to control copter with RC controller?

bkueng, 2017-03-17:
Where are you running the px4 binary? Just to clarify: you should be running it on the RPi, and build the specific target for that, as documented here: https://dev.px4.io/starting-building.html#raspberry-pi-23-boards

L4ncelot, 2017-03-17:
Well I've built the firmware on **RPi** using `make posix_sitl_default`. I didn't build the **posix_rpi** target because I wanted to be able to control quadcopter in Gazebo. And I don't know whether the px4 simulation also gets build or not.

So to clarify, I'm running px4 firmware on my **RPi** which is connected through ethernet to my PC where the **Gazebo** simulation runs.

bkueng, 2017-03-20:
That sounds correct then. You might want to check for some RPi specific ifdefs (`__PX4_POSIX_RPI`) if you use the `sitl` target (I don't see anything in particular, except for the preflight checks that are disabled on RPi).
The alternative would be that you build posix_rpi, and then instead of the simulator, you connect gazebo to the mavlink module. That's how jMAVSim does it.

L4ncelot, 2017-04-03:
Ok, so finally after some time spent on reading the source code of px4 firmware, I've figured it out. The parameters for px4 are all stored in `parameters.xml` file. There are parameters such as **RC_MAP_ROLL** or **RC_MAP_PITCH** which are by default set to 0 and therefore not assigned.

I just created configuration file in `posix-configs/SITL/init/lpe` folder and set RC controller mappings like this:
```
param set RC_MAP_ROLL 4
param set RC_MAP_PITCH 3
param set RC_MAP_YAW 1
param set RC_MAP_THROTTLE 3
```

Now I'm able to control the copter with RC controller after **arming** it using **MAVROS**.

closed 2017-04-03 by L4ncelot