PX4/PX4-Autopilot #6836: EKF HIGH IMU GYRO BIAS

benguru, 2017-03-16:
I have been stuck for a while with the above mentioned preflight fail.
Re-calibrating the system doesn't seem to do the job, nor increasing the COM_ARM_EKF_GB value to its maximum.

I am running px4 on the pixracer platform. It has never boot up without showing the preflight error.
Is it my board that is faulty ?

LorenzMeier, 2017-03-16:
Most likely since we're not seen it across the board population. Do you know how to run something in a shell? That would allow to diagnose it more.

benguru, 2017-03-16:
Unfortunately I forgot my JST cable at school...
But I would take any suggestion on the things that I might check ? In order to come back with a more detailed diagnosis.

benguru, 2017-03-16:
I have checked the gyro output on the analyze widget of QGC and they "seem" fine though...

benguru, 2017-03-19:
It seems to be my custom airframe that is causing the issue...
Is it normal that the horizon and heading widgets don't work on qgroundcontrol when using a custom airframe ?

I am trying to create an airframe the proper way for a foiling boat, its control surfaces are very similar to a fw. But just copying the parameters (rc.fw_apps -> rc.boat_apps and rc.fw_defaults -> rc.boat_defaults) adding a mixer and an airframe seems to have messed up everything...

Any clues on where to find a step by step instruction to doing such things the proper way ?

benguru, 2017-03-19:
Or is there a way to disable preflight checks so I can continue to learn how everything works without this issue always bugging me :)

closed 2017-03-24 by benguru

tstellanova, 2017-05-03:
FWIW I'm encountering this same problem with a brand new Pixracer and the latest stable px4 firmware.  The airframe I selected is the standard DJI Flamewheel F450 , nothing custom.

TrishantRoy, 2018-03-30:
Were you able to solve the problem? @benguru  @tstellanova 
I am facing the same issue.

Thanks in advance.

jscatena88, 2018-06-17:
Same issue here on a pixracer. Anyone figure this out or should I try to RMA the board?

dagar, 2018-06-17:
Please open a new issue and post your log.