PX4/PX4-Autopilot #6837: att_pos_mocap gives no messges

bhairavmehta95, 2017-03-16:
I am using the mavros mocap pose estimate plugin to feed Px4 VICON data as state estimation.

Yet when yet I use the listener att_pos_mocap command in the PX4 shell, nothing shows up.

I have made no modifications to the code, so I was wondering if someone could help me out.

karanchawla, 2017-03-16:
I am having quite the opposite issue, though I'm receiving `att_pos_mocap` msg on the topic I am unable to switch to position control for inav. Any clues about this? 
As for your problem how are you sending `att_mocap_pos` with PX4, indoor fake GPS?

mhkabir, 2017-03-17:
@bhairavmehta95 Does the mavros instance connect with PX4 correctly? Is this on Snapdragon?

mhkabir, 2017-03-17:
And everyone having issues please post logs @karanchawla @bhairavmehta95.

closed 2017-04-02 by mhkabir