PX4/PX4-Autopilot #6839: Pixhawk does not emit any PWM for camera trigger

FPVflyer, 2017-03-17:
Hi guys,
I recently built and set up an FX-79 Buffalo, it's equipped with 3DR Pixhawk hardware, the good old big one, and it fly very well.

I installed a Sony NEX-7 camera onboard for aerial mapping, and I just uploaded a mission generated using QGC. The drone follow precisely the grid without any problem, but for some reason the trigger is not activated.

In the QGC section for camera, it was by default set as PWM emitted on port AUX 5 and 6.
Considering I was need just 1 port (the NEX camera is activated by IR sensor, which work with PWM) to trigger the camera, I disabled the AUX5 and let the AUX6 on. For some reason the AUX6 has never emitted a signal during the whole flight, and the log confirm, nothing came out on port AUX6.

Any chance to set a PWM output for triggering a camera on an MAIN port instead of AUX to make it work?

Thanks

Dam

NaterGator, 2017-03-20:
Please see #6092 and #6194

FPVflyer, 2017-03-30:
Hi, thanks for your help.
I tried as suggested to move the AUX to 3 and 4, I tested again but sadly it still doesn't take picture. Log here: http://logs.uaventure.com/view/LKPD4qmviHBbQzmYEp7Kz3

Please, can someone help me fix this? I really need this working as I'm using this frame for my professional employ, and I don't want to switch to APM.

Best Regards

Damiano

mhkabir, 2017-04-03:
Closing in favour of #6961. Thanks for reporting.

closed 2017-04-03 by mhkabir