PX4/PX4-Autopilot #6843: Bad position holding on some location on earth (simulator)

matanhavi, 2017-03-18:
I have tested the simulator on some random locations around the world, in order to find anomalies.
I found.
for example:
export PX4_HOME_LAT=-30.397742
export PX4_HOME_LON=-10.545594
export PX4_HOME_ALT=1000.0
make posix_sitl_ekf2 gazebo

at first I thought it related to WIP: mc_pos_control auto handling #6810.
but testing it on master also produce similar behavior

just do a takeoff, you will see that the mc start to do pretty bad toilet wash, and other strange staff
[image]

anassinator, 2017-03-19:
We've had a lot of similar problems in Montreal with fixed wing SITL that have made SITL almost unusable.

```bash
export PX4_HOME_LAT=45.436399
export PX4_HOME_LON=-73.945387
export PX4_HOME_ALT=31.293
make posix_sitl_ekf2 gazebo_plane
```

For convenience, generated logs:
Fixed wing as above with a simple mission: http://logs.px4.io/plot_app?log=1b7e618b-30db-4b88-8471-4e9a8f7693ec
Fixed wing as above with a more complicated mission: http://logs.px4.io/plot_app?log=d2f72b80-fba5-4095-9950-6debeed8f3db
MC at OPs coordinates: http://logs.px4.io/plot_app?log=ab5f3fd2-5a33-4f18-99f6-03f74421a93b

Note that in both cases, the random jumps coincide with a `[lib__ecl] EKF GPS fusion timeout - reset to GPS` warning, and that there seems to be regions that always cause a reset and others that are more stable.

matanhavi, 2017-03-19:
@anassinator  I also see the EKF GPS fusion timeout - reset to GPS warning.
I think the issue might be related to the hard codded magnetic field:
[hardcodded mag](https://github.com/PX4/sitl_gazebo/blob/f6a116d7741e70af729f2513ffbf9cc036bd291f/src/gazebo_mavlink_interface.cpp#L447)
although even when I try to change it to the new home position magnetic field, it still not perfect

dagar, 2017-03-25:
We could expand the automated mission tests to randomize the start location.

matanhavi, 2017-03-26:
I found the reason of the bug.
The declination calculation of the simulator should be reversed.
If I reveret the sing at [declination calc](https://github.com/PX4/sitl_gazebo/blob/f6a116d7741e70af729f2513ffbf9cc036bd291f/src/gazebo_mavlink_interface.cpp#L664) its seems that it behave better.
I am not sure why.
- More impotent, are all the calls to get_mag_declination() are on the right sign all over the code?
- on the simulator, what is the initial heading? 90degs? What mag_d_ on GazeboMavlinkInterface::Load should be? the real magnetic on the home pos? just random number with 0 on Y (Its seems to work most of the times)?

LorenzMeier, 2017-03-26:
The whole statement is wrong. I think it got broken recently. There is declination and inclination. Declination should be a rotation around yaw. Inclination is something entirely different and is the Z component of the vector.

To fix this, change this to:

x, y, z
0.4, 0, 0.6

Rotate this fixed vector by negative declination.

matanhavi, 2017-03-26:
@LorenzMeier :
> ... the 3D vector is ok 

Do you mean by that that my observation about the sign is wrong?
I think that in that point the simulator create new magnetic reading, "simulate" the magnetic vector, and that why sign is wrong.

>  The drone should face exactly east.

where can you see that?

I tested on Seattle (declination of +15) and on other random locations (-10,-40, declination -22, for example) the fix works.
On both, the heading is not east without the fix.
On both, the the simulator behaves ok. With the fix, and very wrong without.
I think that because Zurich has onlt 2.2 Declination, you didn't see that much.

There is some issue regarding overshoots, and slide to the side, but I think its just how the navigation works now.

I can easily do the PR, with a fix. I just want to make sure that its the correct fix.

I couldn't find data about 'RotateVectorReverse()' in order to define it the call for it is wrong.

I can also say that the section  initial magnetic vector [hardcodded mag](https://github.com/PX4/sitl_gazebo/blob/f6a116d7741e70af729f2513ffbf9cc036bd291f/src/gazebo_mavlink_interface.cpp#L447) doesn't need to be changed. Basically its random numbers (Y must stay 0) 
still, on some cases there are very strange behaviors when I change it, can't tell why.

dagar, 2017-04-11:
Was this fixed?

anassinator, 2017-04-13:
@matanhavi @dagar @LorenzMeier 

I can confirm that flipping the sign on the declination works in Montreal (declination -14) and Zurich as well. Using `RotateVector()` instead of `RotateVectorReverse()` also fixes it. I could be wrong, but I believe the second fix makes more sense as I've tested out on a few places and the declination's sign seems to be correct:

| City | Expected Declination (deg) | SITL Computed Declination (deg) |
| ---- | --------------------------- | --------------------------------- |
| Montreal | -14 | -15 |
| Seattle | +15 | +16 |
| Zurich | +2 | +0.7 |

Also, I don't understand why the declination quaternion needs to be reversed in the first place?

On a side note, I also believe that `mag_d_`'s z component should be positive as per WMM2015 and that this was simply overlooked when switching from the ENU frame to NED in PX4/sitl_gazebo#62. But I am not sure if I'm reading this properly and if this matters at all. The effects of the x and z vectors seem negligible.

matanhavi, 2017-04-14:
@anassinator 
you can see my notes on the prev post,
I think that in that point the simulator create new magnetic reading, "simulate" the magnetic vector, and that why sign is wrong.
I had some trouble to make PR to fix the sign, can you do it?

anassinator, 2017-04-14:
@matanhavi I don't completely understand what you mean by that to be honest.

From my understanding, we've hardcoded the magnetic field of Zurich in the NED frame but discount the eastern component. And then we want add the declination's effect to the magnetic field as a rotation around the z axis. Now, I've verified that the declination is indeed of the correct sign above, so the issue is how we're applying it.

My hypothesis is that we should not be doing `RotateVectorReverse()` - which is the equivalent of yaw-ing the opposite direction (i.e. subtracting the declination instead of adding it). I think this is also due to the change from ENU to NED in this portion of the code a while back.

I'll make a PR now with the `RotateVector()` fix, but I can change that to flipping the declination's sign if you think that's more appropriate in the review.

PR #8545 (2018-01-01): Update SITL Gazebo for magnetic declination fix
Just bringing in the changes needed to fix #6843.
closed 2017-12-31 by priseborough